darthcloud/BlueRetro #4: BT stop rumble packet drop? Rumble dont stop sometimes

darthcloud, 2020-05-23:
Maybe need to add so robustness in rumble code. Send stop twice??

kieranallen, 2021-09-17:
Hi, I have noticed that rumble is pretty consistent with DS4 but I have a DS3 and that gets trapped in a rumble constantly - did you ever revisit this issue? Thanks

darthcloud, 2021-09-18:
@kieranallen Do you have specific game(s) were its easy to reproduce? I'll look into it.

kieranallen, 2021-09-18:
Sure - I was able to get it to happen very consistently with Ape Escape on PS1. When you do a full jump and land (hold R1), there is a little rumble when you hit the ground, and that was making my DS3 get stuck in a rumble.

darthcloud, 2021-10-05:
@kieranallen I could not reproduce with the jump in Ape Escape but I got a very late DS3 model so could be different.

Can you give this firmware a try?
[v1.0-wip1.zip](https://github.com/darthcloud/BlueRetro/files/7281926/v1.0-wip1.zip)

This should limit the rumble duration and prevent infinite rumble.

kieranallen, 2021-10-05:
Thanks - I've tried it out. It does help, as it eventually shuts off the rumble but the rumble still regularly triggers for too long. I got another DS3 the other day and it also behaves the same way.
[Here's a video showing the behaviour](https://www.youtube.com/watch?v=G8Qlg_G7nsc)

darthcloud, 2021-10-05:
What is the model number and FCC ID of both controller on back sticker?

kieranallen, 2021-10-05:
I am in the UK so there are no FCC IDs.
The model number is CECHZC2E for both.

darthcloud, 2021-10-11:
[v1.0-wip2.zip](https://github.com/darthcloud/BlueRetro/files/7325555/v1.0-wip2.zip)

Reproduced with a CECHZC2U easily and this FW fix it.

Btw my other controller was a CECHZC2UB1 and I couldn't reproduce with that one.

Let me know if this look good for you @kieranallen

darthcloud, 2021-10-12:
Actually better try this more polish version @kieranallen 
[v1.0-wip3.zip](https://github.com/darthcloud/BlueRetro/files/7326237/v1.0-wip3.zip)