darthcloud/BlueRetro #12: Original SIXAXIS PS3 controller disconnects after few seconds

Demonic098, 2020-09-24:
Original sixaxis ps3 controller the one without rumble  in clear case and without dualshock branding disconnect after few seconds. It workfs fine with PS3 console and other DS3 controllers working fine with BlueRetro.
`
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_CONN_REQUEST
# bt_hci_cmd_accept_conn_req
# Page dev: 0 type: 0 bdaddr: 00:19:C1:5F:71:AD
# BT_HCI_EVT_CMD_STATUS
# BT_HCI_EVT_ROLE_CHANGE
# BT_HCI_EVT_CONN_COMPLETE
# dev: 0 acl_handle: 0x0080
# bt_hci_cmd_le_set_adv_disable
# bt_hci_cmd_remote_name_request
# BT_HCI_EVT_CMD_COMPLETE
# BT_HCI_EVT_CMD_STATUS
# BT_L2CAP_CONN_REQ
# BT_HCI_EVT_REMOTE_NAME_REQ_COMPLETE:
# dev: 0 type: 1 PLAYSTATION(R)3 Controller
# BT_L2CAP_CONF_REQ
# BT_L2CAP_CONF_RSP
# BT_L2CAP_CONN_REQ
# BT_L2CAP_CONF_REQ
# BT_L2CAP_CONF_RSP
# bt_hid_ps3_init
# bt_hid_ps3_init_callback
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_L2CAP_DISCONN_REQ
# BT_L2CAP_DISCONN_REQ
# BT_HCI_EVT_DISCONN_COMPLETE
# DISCONN from dev: 0
# bt_hci_cmd_periodic_inquiry
# bt_hci_cmd_le_set_adv_enable
# BT_HCI_EVT_CMD_COMPLETE
# opcode: 0x0403 error: 0x0C retry: 0
# BT_HCI_EVT_CMD_COMPLETE
ERR
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_INQUIRY_COMPLETE
# BT_HCI_EVT_INQUIRY_COMPLETE
`

darthcloud, 2020-09-24:
I ordered a couple "Sixaxis only" and will take a look. Thanks!

Demonic098, 2020-09-24:
I don't know if it could help but http://downloads.arduino.cc/libraries/github.com/jvpernis/PS3_Controller_Host-1.0.0.zip arduino based library working fine with DS3 and Sixaxis PS3 controllers.

Demonic098, 2020-09-24:
Also if the ramble feature is not enabled in BlueRetro config (Tested on nintendo GameCube) any DS3 gamepad not receiving port number indication. It Keep blinking all 4 leds as if it is not connected. But if ramble feature is enabled after first ramble executed it set port number.

Demonic098, 2020-09-24:
In sources BlueRetro/main/bluetooth/hidp_ps3.c there is function  bt_hid_ps3_init_callback which i belive have to init gamepad and set gamepad port number but it never set it.
```
https://github.com/darthcloud/BlueRetro/blob/master/main/bluetooth/hidp_ps3.c
static void bt_hid_ps3_init_callback(void *arg) {
    struct bt_dev *device = (struct bt_dev *)arg;
    struct bt_hidp_ps3_set_conf set_conf;
    memcpy((void *)&set_conf, ps3_config, sizeof(set_conf));
    set_conf.leds = (bt_hid_led_dev_id_map[device->id] << 1);

    printf("# %s\n", __FUNCTION__);

    esp_timer_delete(ps3_timer_hdl);
    ps3_timer_hdl = NULL;

    bt_hid_cmd_ps3_bt_init(device);
    bt_hid_cmd_ps3_set_conf(device, (void *)&set_conf);
}
```

Demonic098, 2020-09-24:
I've found found a solution. If i connect DS3 to USB port right after BT pairing process and disconnect USB after 10 seconds gamepad stays connected.

Demonic098, 2020-09-24:
Maybe something wrong with my gamepad.I do not have other sixaxis to test it more.

Demonic098, 2020-09-24:
It works exactly the same if I connect it to my PC, it is definitely a defective gamepad. Sorry to bother you.

closed 2020-09-26 by Demonic098