darthcloud/BlueRetro #13: Not receiving port number indication on DS3

Demonic098, 2020-09-24:
if the ramble feature is not enabled in BlueRetro config (Tested on nintendo GameCube) any DS3 gamepad not receiving port number indication. It Keep blinking all 4 leds as if it is not connected. But if ramble feature is enabled after first ramble executed it set port number.

Demonic098, 2020-09-24:
In sources BlueRetro/main/bluetooth/hidp_ps3.c there is function  bt_hid_ps3_init_callback() which i belive have to init gamepad and set gamepad port number but it never set it.
```
https://github.com/darthcloud/BlueRetro/blob/master/main/bluetooth/hidp_ps3.c
static void bt_hid_ps3_init_callback(void *arg) {
    struct bt_dev *device = (struct bt_dev *)arg;
    struct bt_hidp_ps3_set_conf set_conf;
    memcpy((void *)&set_conf, ps3_config, sizeof(set_conf));
    set_conf.leds = (bt_hid_led_dev_id_map[device->id] << 1);

    printf("# %s\n", __FUNCTION__);

    esp_timer_delete(ps3_timer_hdl);
    ps3_timer_hdl = NULL;

    bt_hid_cmd_ps3_bt_init(device);
    bt_hid_cmd_ps3_set_conf(device, (void *)&set_conf);
}
```

darthcloud, 2020-09-29:
Reproduced the issue with a few `new` controllers I got:
SIXAXIS FCCID: AK8CECHZC01
DS3 FCCID: AK8CECHZC2

My own DS3 controller work fine:
DS3 FCCID: AK8CECHZC2UB1

Will dig into it later this month. Thanks for reporting.

Demonic098, 2020-09-29:
All my working controllers have the same FCCID: AK8CECHZC2.

darthcloud, 2020-09-29:
[v0.7.1_wip.zip](https://github.com/darthcloud/BlueRetro/files/5299892/v0.7.1_wip.zip)

Please try this FW, should fix it.

Demonic098, 2020-09-29:
Yes, that solved the problem.

Demonic098, 2020-09-29:
When it will be added to the sources ? I need to add spiffs support.

commit d850feb2 (2020-09-29): [PS3] Fix again PS3 timing for earlier controller version

Fixes #13
closed 2020-09-29 by darthcloud