darthcloud/BlueRetro #17: is hardware universally required?

drunkscientist, 2021-02-07:
so some consoles clearly need some elaborate communication protocols, but im only ever planning to use it with an N64 and with one controller(my brother is a purist, gotta be og... idgaf). can i use any random esp32 dev board and hook up [gnd, vin, and io19](https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-Cables-Build-Instructions#nintendo-64-adapter-cable)?

darthcloud, 2021-02-07:
No but without sdcard card everything will be on default settings and so the adapter will be in auto-detection mode.

So as noted in the cable schematic:
https://github.com/darthcloud/BlueRetroHW/blob/master/DIY/N64.pdf

Very important for (I39/VN) to be connect to VIN (3.3V) for proper system detection. 

Also you need to GND the other 7 auto detect inputs. Its not obvious in the schematic since you need to crosscheck the DB25 pin with the DIY schematic to gets the ESP32 IOs:
https://github.com/darthcloud/BlueRetroHW/blob/master/DIY/BlueRetroDIY.pdf

In any case GND those 7 IO pins:
 21, 22, 25, 32, 33, 34, 35,

This will make sure the adapter always detect a N64.

closed 2021-02-07 by darthcloud

drunkscientist, 2021-02-07:
awesome, thank you!

drunkscientist, 2021-02-07:
but wait, is I39 not the same as gpio22? which gpio gets pulled high?

darthcloud, 2021-02-07:
No its different, it could be label VN, IO39 or I39 on your board.

drunkscientist, 2021-02-08:
found it, mh-et live mini kit labels it as SVN. thanks again!