darthcloud/BlueRetro #19: Trying to get ps2 to work with BlueRetro

ZiroGX2, 2021-03-31:
I have an ps2 directly wired to a esp32 devkit v1 with wires and have tried to get paired ps4, xbox one, and wii controllers to function. They pair to the device fine but dont input. I am using the ps2 firmware and only have one set of controller wires hooked up. I have tried messing with the advanced config but to no avail. Is there anything I am doing wrong?

darthcloud, 2021-03-31:
Can you provide picture of the DevKitC wiring were I can see all 8 wires easily. If wireless ok then power is probably fine, its likely one of the connection DTR, RXD, TXD, DSR or SCK. 

You followed instructions from https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-Cables-Build-Instructions#psx--ps2-adapter-cable (the part on setting P2 pin to 3V3 if unused)?

Which firmware version are you using?
Which firmware variant are you using (filename) ?

What is the SCPH model number of your console?

darthcloud, 2021-04-12:
Reopen if further help required.

closed 2021-04-12 by darthcloud