darthcloud/BlueRetro #20: PS2 rumble speed control on Dualshock 4

ttyborg, 2021-04-05:
Hi!

I have built an adapter out of a devkitv1 using the 0.14 release, and it works perfectly with my DS4.
I managed to turn on the rumble through the web interface, and it works, but it seems the rumble motors don't work independently of each other, and they only work at a set value, and don't have speed control. I tested this on a PS2 with the "Pad test" homebrew app (https://www.ps2-home.com/forum/viewtopic.php?t=5483). I also have a Brook PS2 Super Converter, with it the rumble motors can work independently, and have speed control. I don't know if I am missing a config or something.

Thanks for your help.

darthcloud, 2021-04-05:
You are correct, dual motor and speed control are not yet implemented. It something that will require redesign later.

I will eventually fix it but not in the short term.

BTW, thanks alot for the github sponsorship, I really appreciate it!!

ttyborg, 2021-04-05:
I see, thanks for the quick reply.

For the sponsorship, it's the least I can do. Keep up the good work.

patters-syno, 2021-11-23:
I use a PSUnoRetro device (from willsconsolemodifications.co.uk) running BlueRetro, with a PlayStation 2. I have enabled BlueRetro rumble via the web config, and it works for PS2 games.

I have noticed that there's no rumble at all for Porsche Challenge - a PS1 game working on the PS2's backwards compatibility. Using a wired DualShock 2 pad rumble does indeed work, so it might logically be expected to work with BlueRetro.

However it does feel as if only the right-hand vibration motor is used when using a real DS2 pad. From memory of owning this game back in the day, and the PS1 Dual Analog controller, I believe this game was one of the first to support that type of controller.

Weirdly the Dual Analog controller had its rumble motor removed for the US and European markets, but did have a single rumble motor in Japan (https://en.m.wikipedia.org/wiki/Dual_Analog_Controller). So it would make sense that this game supports only one vibration motor, even if the feature then works with the later DualShock and DualShock 2 controllers, which are enhancements of that original design.

Is it the case that BlueRetro only activates the DS4 vibration motors when output for the left-hand motor is requested, and perhaps ignores requests for the right-hand motor? Or is there some other issue here?

darthcloud, 2022-01-26:
@patters-syno While only one motor from the PSs BT ctrls is used both motor command on the PSX/PS2 side should activate it. I will need to take a look. Sorry I missed you report in november.

patters-syno, 2022-01-27:
Thanks for the confirmation that it ought to work. I wasn't sure, that's why I hadn't raised it as a distinct issue. I can do this now if you'd rather, for tracking purposes.

darthcloud, 2022-01-27:
@patters-syno yes please open a different issue, I'll keep this one to track the BT side. Just copy paste your first post.

mario-64, 2022-07-28:
> You are correct, dual motor and speed control are not yet implemented. It something that will require redesign later.
> 
> I will eventually fix it but not in the short term.

Hi darthcloud. I was wondering if this feature is on your roadmap now or still no current plans to add it?

mario-64, 2023-01-05:
> You are correct, dual motor and speed control are not yet implemented. It something that will require redesign later.
> 
> I will eventually fix it but not in the short term.
> 
> BTW, thanks alot for the github sponsorship, I really appreciate it!!

> You are correct, dual motor and speed control are not yet implemented. It something that will require redesign later.
> 
> I will eventually fix it but not in the short term.
> 
> BTW, thanks alot for the github sponsorship, I really appreciate it!!

@darthcloud This post was nearly two years ago. Do you still plan on implementing proper DualShock rumble support? It’s the only thing holding me back from using BlueRetro with PlayStation consoles. Thanks

ButtonBasher92, 2024-01-15:
Hoping this gets implemented in the next major update, ps2 support would then be complete and perfected.

mario-64, 2024-01-15:
I’m holding off purchasing any BR PlayStation products until it is. This has been a known issue for nearly 3 years now.

darthcloud, 2024-01-15:
Locked since 100% of the comment adds nothing useful.

I'm aware of it, but on my priority list this is simply low.

This will be done when I feel like working on it or someone else send a PR.

ButtonBasher92, 2024-05-29:
> Can someone test this beta, this add dual motor and variable force support for Playstation and Dreamcast
> 
> https://github.com/darthcloud/BlueRetro/actions/runs/9286878322

So far working great in a few games that I have tested.
I am using PS3 controllers
I have noticed my PS3 controller with model number CECHZC2U has a delay with the vibrations though
CECHZC2E works fine, no delay.

DJm00n, 2024-05-29:
@darthcloud Did a test pass with the PadTest tool on PS2 Fat with [HW1 BlueRetro from Aliexpress](https://www.aliexpress.com/item/1005004662621894.html):

>To test vibration, hold 'triangle', then move the left joystick down for the strong motor (that seems to have 96 positions), or press up and down on the directional pad for the weak motor (that has 2 positions).

Tested with two DualShock4's connected at the same time, two DualSense's connected at the same time, one DualShock4 and a second DualSense.

Works as it should, but with a bug - **the vibration is only there when two controllers are connected to the BlueRetro**.

darthcloud, 2024-05-29:
@DJm00n I did all my test with the PadTest on fat PS2 too and I had only one controller connected.

I also tested the stadia controller too and it worked fine.

TBH those adapter look poorly build (in addition of them violating the license by using the project name to name themself.)

darthcloud, 2024-05-29:
@DJm00n I reproduced your issue. Something is wrong when rumble is enable on both output 1 & 2 and you got only one controller.

Disabling rumble on output 2 make output 1 work fine.

I will look into it.

darthcloud, 2024-05-29:
@DJm00n That should fix it:
https://github.com/darthcloud/BlueRetro/actions/runs/9292253020

DJm00n, 2024-05-29:
@darthcloud works like a charm! thank you!

PS: Still no vibartion on Stadia Controller at all.

mario-64, 2024-05-29:
@darthcloud Just want to say a huge THANK YOU for implementing this! Due to work, I can't test right now but can this weekend.

darthcloud, 2024-05-30:
This one should fix RE4:
https://github.com/darthcloud/BlueRetro/actions/runs/9304053152

darthcloud, 2024-10-06:
Fixed in v24.10:
https://github.com/darthcloud/BlueRetro/releases/tag/v24.10

closed 2024-10-06 by darthcloud