darthcloud/BlueRetro #24: analog control on PS1

Micha213, 2021-04-16:
I discovered a problem with the analog control on PS1.
In the game Resident Evil 3 (I tested the EU version, loaded with the xStation) the analog stick to the right sometimes does not work. Analog stick to the left, up and down works without any problems. It affects all menus: title screen menu, option menu and inventory menu. But I can control the characters in the game without any problems.
I had noticed a similar problem in the PSIO menu.  Here some directions of the analog stick did not work either. So far these were the only two situations with which the analog control caused problems. Other games like Silent Hill, Metal Gear had no problems.
I tested the firmware v0.13-psx_ps2_spiffs, v0.15.1-psx_ps2_spiffs and v0.15.1-universal_spiffs each with an Xbox One S and PS4 controller.

darthcloud, 2021-04-16:
Maybe your BT controller dont go full right?

In the web cfg if you increase the PAD_LX_RIGHT maximum to something like 130% as a test does it work?

darthcloud, 2021-04-16:
I reproduced easily with RE3 and my DualSense ctrl, Maximum value will vary from ctrl to ctrl but I will double check to make sure the scaling work properly.

Meanwhile setting axis to 110% fix the issue for me:
[image]

Micha213, 2021-04-16:
Thanks. I just tested it with RE3 and it works for me at 110% too.

darthcloud, 2021-04-17:
[v0.15.1_WIP3.zip](https://github.com/darthcloud/BlueRetro/files/6329726/v0.15.1_WIP3.zip)

Try this (put back config to 100%) and let me know if it look ok for you. 

Tested with PS3/PS4/PS5, Xbox One S and WiiU pro ctrl.

DualShock controller reach maximum value in either direction (0x00, 0xFF)
a little bit before reaching the physical limit. This allow to always reach
maximum value event on wearout joysticks.

Some games (like RE3 in menus) took the poor design decision to respond to joystick movement
in a digital way by expecting exactly 0x00 or 0xFF! 0x01 or 0xFE is not enough!!

This look to be quite common assumption on PSX games so I tweaked the maximum value so
that the scaling by default is likely to saturate without loosing significant range.

This is a trade off. This won't help for extremely wearout BT controllers or generic
HID controller with weird maximum value. Those will need to adjust the maximum value
via the web configuration.

closed 2021-04-20 by darthcloud