darthcloud/BlueRetro #49: Copyable version of the Web Config Mapping reference

vaderag, 2021-06-30:
The [sheet ](https://docs.google.com/spreadsheets/d/e/2PACX-1vRln_dhkahEIhq4FQY_p461r5qvLn-Hkl89ZtfyIOGAqdnPtQZ5Ihfsjvd94fRbaHX8wU3F-r2ODYbM/pubhtml) that is published doesnt allow you to edit/copy but would like to trim down - is it possible for you to share a version that can be copied and edited so I can create a reference for only the systems I use?

Thanks!

darthcloud, 2021-06-30:
You can just select all, copy and paste in your own spreadsheet.

closed 2021-06-30 by darthcloud

vaderag, 2021-06-30:
> You can just select all, copy and paste in your own spreadsheet.

It seems to lose all the formatting when I do that - think due to the merged cells in column one 
You can't just copy a column either

darthcloud, 2021-06-30:
work fine for me with both google sheet and excel. Formatting is not loss.