darthcloud/BlueRetro #56: GameCube Digital Triggers Analog Merge

shimmytoast, 2021-08-02:
Love the project just got a DIY unit built up and it's running great. I'm using the internal memory and the latest .19 binary universal.

I ran into what I believe is a bug when I try to select the preset profile for GameCube Analog and Digital Triggers Merge. Once I do this and power cycle the unit I can no longer access the advanced settings through the web browser and I no longer get any output to the GameCube. I also tried to use the GameCube specific binary and saw the same issue.

If I use the OTA update to re-download the main binary everything starts working like normal again. Good looking out on that too BTW because I can't access my USB port easily once my DIY unit is fully assembled.

darthcloud, 2021-08-03:
Fixed in v0.19.1
https://github.com/darthcloud/BlueRetro/releases/tag/v0.19.1

Thanks again!

closed 2021-08-03 by darthcloud

shimmytoast, 2021-08-04:
Awesome glad I could help hell yeah!

Thank you also!