darthcloud/BlueRetro #69: Two identical controllers seem to both connect to same channel

vaderag, 2021-09-19:
I'm not 100% certain why this is happening but I have two identical 8bitdo controllers and initially thought one must be connecting to something else, but I've been round and turned off all Bluetooth I can find and now can only assume that they are both connecting on the same channel

Any ideas why this might be?

darthcloud, 2021-09-19:
Do they work properly with BlueRetro one at a time?

Can you get the debug logs: https://github.com/darthcloud/BlueRetro/wiki/Getting-BlueRetro-debug-logs-via-Serial-port-Windows-10

vaderag, 2021-09-19:
Yep, one at a time is good. Will check the logs when I have a moment

vaderag, 2021-09-19:
Okay, so I've worked out what it is, and there is an issue but not the same as I thought

The controllers in the pic are all connected
 Notice how there are two ones and one 3. Turns out that 8bitdo #2 is connected but not receiving(?) the correct status assignment. So it always shows as P1
I've tried connecting in all positions and irrespective what player it's assigned to it shows #1
Also connection mode doesn't make a difference (except to the button layout), still shows as P1 in Switch or Xinput modes

[image]

darthcloud, 2021-09-24:
This is likely just a cosmetic LED issue, you should have one being P1 and the other being P2

vaderag, 2021-09-25:
> This is likely just a cosmetic LED issue, you should have one being P1 and the other being P2

Oh it is just cosmetic - but it's quite confusing given other controllers seem to connect with the correct # designation.

darthcloud, 2022-01-17:
Closing this, 8bitdo ctrl need to be used in Xinput mode which provides no player LED indicator.

If official SW ctrl where also concerned by this issue it's likely  a manifestation of issue #146

closed 2022-01-17 by darthcloud