darthcloud/BlueRetro #75: Incorrect button inputs on GC using Xbox One S controller

TheMikeBachmann, 2021-10-03:
I am using an Xbox One S controller (on gamecube using universal fw 1.0) and some of the button inputs are mapped incorrectly even when using the Default Gamepad Only preset. 

Here is a list of buttons on the Xbox One S controller and how they are registering in a controller test application:

Xbox LB = GC Y
Xbox Menu and Share = GC Z
Xbox X = nothing
Xbox B = GC X
Xbox Y = GC B

Neither Bumper register as Z as they should, and there is no start button. (You'd be surprised how many games **require** you to press start to begin playing.) Both sticks register correctly, the DPAD registers correctly, and the triggers register correctly. 

Output config mode is set to Gamepad, though behavior is the same in GamepadAlt mode.

darthcloud, 2021-10-03:
Is that specific to v1.0, is v0.19.1 ok?

What is FW version of Xbox ctrl (you can check in the xbox accessories win 10 app)

TheMikeBachmann, 2021-10-03:
It's currently on 3.1.1221.0

I'll update it and see if that changes anything.

TheMikeBachmann, 2021-10-03:
I have not tried 0.19.1 blueretro FW, this is my first time using it.

TheMikeBachmann, 2021-10-03:
Updating the controller firmware fixed the issue. Thanks!

darthcloud, 2021-10-03:
Awesome, thanks for reporting original FW version, that's good information!

Btw for gamecube you probably want to use the gamecube merge trigger preset, to emulate the GC triggers end button.

TheMikeBachmann, 2021-10-03:
Good call! I'll give that a shot. Thanks for the late night support!

closed 2021-10-03 by darthcloud