darthcloud/BlueRetro #91: Variouse Issues with Xbox one to GC adapter

dur75945, 2021-11-08:
So few things, but first I'll go over my setup I have an ESP32 Devkitv1, a wii running homebrew, (just so I can run Super smash bros melee without having a disk, since that is what I am hoping to use this adapter for) and a esp hard wired to a single male GameCube cable. I do remember that the wiring instructions mentioned tying any unused data cables to 3.3v and wiring all of the 5v lines from the GameCube to vin, so I wouldn't be surprised if some of my issues are a result from a lack of power for the ESP.

fisrt, the default pairing is messed up. few things come to mind, the menu and select buttons (I believe that is what they are called, they are the 2 buttons under the home button) are bound to "z", Xbox b is bound to GC x or y (unsure of which), Xbox y is bound to GC b. nothing is bound to GC start, etc. That being said this part could very well be user error.

second is that inputs being registered longer than they actually are being press, I have a video showing this. in smash there is a full hop and a short hop. in the case of bowser, if you let go of jump less than 8 frames after you initially input jump, you short hop, if you are still holding jump 8 frames after you first input it, you full hop. in the video the left bowser is the Xbox controller, the right is a oem GC controller. the Xbox/esp are still registering me pressing jump when I am no longer doing so, about 1/10th of a second. This does not happen when I use tap jump (pushing the analog stick up) to short hop.

finally is an analog stick drift. if I move the control stick from neutral position to slightly down repeatedly. the game starts to register the stick as down even when the stick is actually in neutral position, same happens when doing this in other directions. I didnt notice any issues beyond all these 3 other than the controller taking a while to connect, like 30 or 40 seconds, but that's about it.

attempting to short hop with GC x or y
https://user-images.githubusercontent.com/85606597/140694499-9131d2b2-238e-46cb-8760-95f3961ffc1c.mp4

short hoping with tap jump
https://user-images.githubusercontent.com/85606597/140694513-c5bb56a1-eb46-45a9-848c-35e42656a741.mp4

the stick drift, by the very end I am crouching even when the stick is in neutral position
https://user-images.githubusercontent.com/85606597/140694699-31cde677-2329-4223-a69e-17f93f454098.mp4

darthcloud, 2021-11-08:
To fixup the mapping you need to update your controller with the xbox accessories win 10 app. This might fix other things as well.

dur75945, 2021-11-08:
Thank you for the quick response. I updated the controller, the mapping is kind of working now. top and right face buttons are set to x and y, left face button is set to b and bottom is set to a. The esp isn't holding buttons down beyond when I press them anymore. The stick drift from earlier is no long happening however, there are new stick issues. from what I am seeing is that the esp momentarily disconnects from the Wii, and immediately reconnects. I notice that this is happening a lot when I am trying to recreate the stick drift from earlier, (moving the stick back and forth from neutral position to any direction repeatedly). since the Wii is reconnecting with the esp at a point where the Xbox controller stick is off center, the Wii calibrates around that point  causing the Wii to always be reading the stick off center, at least until the disconnect happens again and the Wii recalibrates. also the esp is always telling the Wii that gc "z" is pressed.

(last minute edit before posting)
the disconnect issue is related to power, if i plug the esp into a usb wall adapter with a micro usb cord, the disconnects stop.

darthcloud, 2021-11-09:
Ok you need to make sure your build is stable first. Try to keep it simple like try to connect only 3 wire 5v, data & GND. The other 3V3 connection are optional in most case. Take a good picture of your build were I can see all connections well.

So while using USB power your only issue is GC Z is always pressed? I will try myself later to confirm.

dur75945, 2021-11-11:
Ok so the issue with z being held was a jammed bumper on my controller. The disconnect issue has just kind of stopped happening, I'll mess around with it more to see if I can recreate it but otherwise thanks for your help, sorry for wasting your time.

closed 2021-11-11 by dur75945