darthcloud/BlueRetro #102: PS1 5501 PU-18 rev. 1-644-537-52 erroneous button inputs

dryja123, 2021-11-29:
Using the PSUnoRetro adapter from Will's Console Modifications.  As soon as I connect the device to my PS1 it starts to spam random button inputs.  I factory reset the device and that did not resolve the issue.  

Reached out to Will and they said that this is a known issue with certain revisions of the 5501 / PU-18.  My board is a PU-18 1-644-537-52.  

Opening an issue for tracking.

darthcloud, 2021-11-29:
Will sent me one of those mobo, will look into it when I receive it.

dryja123, 2021-11-29:
Awesome, he said he was sending one out for you.  Just wanted to open a ticket so it will be searchable for others, if they have an issue.

Batossai84, 2021-12-05:
I have the same problem. pu-18 with ps1digital and xstation installed. I tried 2 different unoretro. One with the original firmware and another with 1.1. They behave the same way, random botton inputs and I could not use a bluetooth mouse either.

marceloMedeiros, 2021-12-07:
> I have the same problem. pu-18 with ps1digital and xstation installed. I tried 2 different unoretro. One with the original firmware and another with 1.1. They behave the same way, random button inputs and I could not use a bluetooth mouse either.

Since i also have a pu18 with ps1digital and xstation and blueretro seems to be working fine on my end, maybe we can figure out whats causing the problem if we compare our setups and use case. In my case, in addition to the aforementioned, my ps1 also has a willconsolemods psu replacement, and im using a nice 12v 5a psu with it.

Can you describe what is your setup exactly? And the steps to reproduce the erractic behavior (i.e. ,what games you are running, etc)

dryja123, 2021-12-07:
> Since i also have a pu18 with ps1digital and xstation and blueretro seems to be working fine on my end, maybe we can figure out whats causing the problem if we compare our setups and use case. In my [...]

When I was speaking to Will about the issue, he was saying that there are different board revisions of the PU-18.  In my case I have a PU-18 1-644-537-52.  Can you confirm that you have the same PU-18 revision?  

For me, the issue happens as soon as I insert the PSUnoRetro.  It's not game specific, even happens at the memory card manager screen.  As soon as you plug the device in it just starts spamming inputs.  

If you could provide your PU-18 revision that would be extremely helpful, since you're not having the issue.

Batossai84, 2021-12-07:
Eur 5502 PU-18 1-644-537-52 serial C5578908 (will psu, will igr, ps1digital, xstation) there are lot of random button inputs.
Usa 5501 PU-18 serial U7559134 (psio) there are less random inputs, but they still exist. Pressing start or if you pulse and mantain a button it is pressed again after some time, while this doesn’t happen on a regular controller.
Eur 7502 pu-20 (psio) everything seems to work fine.
Sorry for not sharing the board of the usa, but I would have to unscrew too many things and I don't have time to do it.

marceloMedeiros, 2021-12-07:
> When I was speaking to Will about the issue, he was saying that there are different board revisions of the PU-18. In my case I have a PU-18 1-644-537-52. Can you confirm that you have the same PU-18 [...]

I wasnt sure what exact revision mine was so had to crack it open. And yep, just confirmed and it seems to be the exact same revision.

Also forgot to mention that on top of everything i already mentioned, im also using a memcardPro.

Finally, i want to clarify that im using a reference Blueretro device, not the PSUnoRetro. And I use just one blueretro for both controllers (with a 2 player cable). 

Im not sure if thats relevant or if the PSUnoRetro is the exact same thing or a custom device with different characteristics compared to the reference design. But i think it might be relevant...since our setup seems to be almost identical, except for the blueretro device itself.

[image]
[image]

dryja123, 2021-12-07:
> I wasnt sure what exact revision mine was so had to crack it open. And yep, just confirmed and it seems to be the exact same revision.
>
> Also forgot to mention that on top of everything i already [...]

Thanks for confirming. I am also using the PS1 Digital, xStation, and Memcard Pro.

Batossai84, 2021-12-07:
[image]

darthcloud, 2021-12-14:
Spent some big buck to get the exact same revision with pre-installed xstation. I tested with the PSUnoRetro Will gave me and I couldn't reproduce the issue.

[image]

I'll do more test over the weekend.

The PSU is original and no HDMI mod.

@dryja123 & @Batossai84 can you list all the mod your respective console have?

I assume you tested without the mem card pro to be sure?

Still waiting to receive Will's one and/or one of Will customer donation.

dryja123, 2021-12-14:
@darthcloud my PS1 is modded with the PS1 Digital and xStation. I’ve tested with and without the memcard pro installed. My first thought was the memcard interfering with the device.

darthcloud, 2021-12-14:
Re-reading the whole thread look like we can't really blame a particular mod mix. Will be easier to troubleshoot when I get the donation. I got a good PU-18 to compare with which is nice.

dryja123, 2021-12-14:
I would love to let you borrow my console but I don’t trust shippers and with the current state of things.  I wouldn’t be able to replace the mods if the console was lost. 

I’m happy to test things for you. Just let me know.

darthcloud, 2021-12-19:
@dryja123  & @Batossai84 I might have a fix but could not test since I didn't receive problematic board yet, but it's tested to work as before on my PSX. Could you give it a try and report back?
[1.2.1_wip.zip](https://github.com/darthcloud/BlueRetro/files/7740343/1.2.1_wip.zip)

With my scope I was able to see some voltage fluctuation on data output:
[image]

This was due to GPIO not being properly configure as open drain. Not I got no fluctuation anymore:
[image]

I guess most console didn't care much about it but some where sensitive to that and would glitch sometimes.

dryja123, 2021-12-19:
Hey @darthcloud I had an opportunity to flash this firmware.  Unfortunately, it did not fix the issue for me. Now when the console turns on the device has a solid white light; indicating an error. I press the reset button (on the unoretro) and the light starts to blink, like it’s pairing. As soon as the light starts to blink it starts to spam button presses.

darthcloud, 2021-12-19:
Did you use v1.2 before? So now at every power on the led stay solid, but  original 1.2 didn't?

Batossai84, 2021-12-19:
I´m busy working right now and I can´t run many test until the holidays are over. Whenever I could I will transplant the controller piece from the 7502 to the 5502 to check if it is a hardware problem there. The unoretro works fine on my 7502. Last time I updated it took 10 hours using my phone, so I can only update while sleeping. My pc doesn´t have the software to identify the unoretro when I connect it via usb.

dryja123, 2021-12-19:
> Did you use v1.2 before? So now at every power on the led stay solid, but original 1.2 didn't?

I used what ever version 0.19.1 and 1.1.1 and didn’t have the issue with the solid white light. I didn’t see the 1.2 release so this was the first time I used it.

darthcloud, 2021-12-19:
@dryja123 when you connect to the adapter via OTA page do you see this line in the page log:
App version: v1.2-3-g0c51667 psx_ps2

If not the flashing failed. You selected the file BlueRetro_psx_ps2_spiffs.bin in the zip right?

dryja123, 2021-12-19:
@darthcloud I didn’t entirely pay attention to what the log said. I did choose that file. I can flash it up again and I’ll pay closer attention to what the log says.

darthcloud, 2021-12-19:
@dryja123  Like just connect to it and check the log, no need to reupload

dryja123, 2021-12-19:
@darthcloud Just hopped on my PC and I left the webpage up from last night.  This is the what the live output said:

`Requesting Bluetooth Device...
Connecting to GATT Server...
Getting BlueRetro Service...
Get Api version CHRC...
Reading App version...
App version: v1.1.1 psx_ps2_spiffs
Init Cfg DOM...
FW upload done. Took: 314.32970000000296 sec
> Bluetooth Device disconnected`

darthcloud, 2021-12-19:
@dryja123 ok weird, mine don't have solid led, thanks for testing. I'll just wait to get the board from Will.

TerminatorSVT, 2022-01-11:
I just noticed the one you couldn't reproduce the problem on is a AQ under the peeled back shield.

darthcloud, 2022-01-11:
Email me to darthcloud gmail.com (replace space with @) or DM me on twitter, we can coordinate this way. I'll pay fedex/ DHL overnight both way.

TerminatorSVT, 2022-01-22:
Fix: power direct from PS1 3.3v instead of 7v + regulator.
[image]

tajunk, 2022-02-11:
> Fix: power direct from PS1 3.3v instead of 7v + regulator. [image]

Thanks for this, but could you clarify a little more on what you did?
Is it just remove the third pin from the right, put it in the middle (just to be off the pad I'm assuming?) and wire it to the regulator?

Just wanted to clarify as I'm decent at soldering but the technical stuff escapes me somewhat lol.

dryja123, 2022-02-11:
> Thanks for this, but could you clarify a little more on what you did? Is it just remove the third pin from the right, put it in the middle (just to be off the pad I'm assuming?) and wire it to the [...]

Reach out to Will’s Console Mods. He said he’s going to issue replacements to those who are having this issue.

tajunk, 2022-02-25:
> Fix: power direct from PS1 3.3v instead of 7v + regulator. [image]

Sorry to bump this again but: I've contacted Will a couple weeks ago now and didn't get a response so it looks like I'm going to have to attempt this fix myself. 
Any chance I could get some clarification from @darthcloud @TerminatorSVT on what's happening here? Is it just changing the pin location from the third on the right to the middle and wiring it to that regulator or am I missing something? Thank you to anyone who could help!

TerminatorSVT, 2022-02-25:
PS has 2x +V pins on the controller port. +3.3v to power controller logic and +7v to power rumble motors.

The old pin location feeds 7v from the rumble and steps it down to 3.3v for the regulator. The tab on the SMD regulator is on the 3.3v output side (note this is different from a typical TO-220 style regulator where the tab is GND). For some reason I haven't dug into this is unreliable. There is prob some offset or 3.3 to 7v rail shorting or something wierd going on.

All this does is relocate the pin to the native 3.3v supply and bypassing direct to the 3.3v regulator output side (regulator is thus unpowered). All the tape is just to make sure the pin doesn't touch anything as its floating only by friction with the new shell hole as there is no solder pad to secure it to.

tajunk, 2022-02-26:
Thanks so much @TerminatorSVT !! Appreciate the explanation and clarification, I just did the fix and it's working perfect now on my ps1!

darthcloud, 2022-04-13:
First off sorry for not actively working on this, but since I don't have HW to reproduce the issue and being now a new father made this important issue less attractive to work on vs adding new features.

Anyway someone suggested this is a BT power issue, so here a FW to try with BT completely off.
[v1.5.1_btoff.zip](https://github.com/darthcloud/BlueRetro/files/8481956/v1.5.1_btoff.zip)

This assume you can reproduce the glitch (random input) in the X-station menu without any BT device connected.
If you do, try this FW and the result will help determine if it related to power draw from the radio or not.

darthcloud, 2022-04-13:
Another suggestion is to place inline 220 ohm resistor between ESP32 inputs and PS pin 2 (TXD aka CMD), pin 6 (DTR aka CS) and pin 7 (SCK). Original controllers got those maybe for a reason!

dryja123, 2022-04-13:
I received the latest revision of the PSUnoRetro from Will, with the 3.3v change, and the replacement unit is exhibiting the same behavior.

dryja123, 2022-04-13:
I ordered up some 220ohm resistors, they’ll be here tomorrow. I can give it a test.

darthcloud, 2022-04-13:
ok cool, I think Will will send you another dongle with those resistor too.

dryja123, 2022-04-13:
Yeah, Will and I had a chat about it. The resistors are cheap and we can have a quick definitive answer.

darthcloud, 2022-04-13:
@dryja123 if you use extension keep the cord between the dongle and console less than 1 feet. Long extension made problems for other people.

darthcloud, 2022-04-14:
I added the resistor this morning, work fine as before on my PU-18. I took 249 ohm resistors as this is what I had on hand.

[image]

dryja123, 2022-04-14:
@darthcloud soldered up the 3 220ohm resistors and I’m still getting random inputs. It doesn’t seem as bad, but it’s still happening.  Just a note, pin 3 does look close to the pins on the ESP but there’s plenty of clearance and it’s not shorting. 
[image]
[image]

dryja123, 2022-04-14:
I wish I could send you the console for debugging but the PS1D is impossible to get. I’d be crushed if the console got lost in transit.

darthcloud, 2022-04-14:
Thanks for taking time to test it.

I think this will ultimately need more serious analysis with a scope on a system with the issue.

The best would be if someone was willing to sell me one.

darthcloud, 2022-04-15:
@dryja123 from what I understand you see random input right away in the xstation menu, right? (no need to connect BT controller).

If it's not too much to ask could you try the no BT FW for BlueRetro?
https://github.com/darthcloud/BlueRetro/files/8481956/v1.5.1_btoff.zip

This will keep the radio off and would help to tell if the issue is due to power draw spike due to RF TX.

dryja123, 2022-04-15:
@darthcloud right away in the xStation menu and also the system BIOS.  If I flash that BT off firmware how do I flash firmware in the future?  Would I need to buy a hardware flasher?

darthcloud, 2022-04-15:
oh yeah I didn't think about this!

Yeah try with the devkit if you can, try first with regular FW to see if you reproduce, then use the no BT FW.

darthcloud, 2022-04-15:
@dryja123 actually holding the boot button 10 sec to factory reset would revert to the original FW the unoretro got ship with.

dryja123, 2022-04-15:
Awesome, I’ll play around with it this weekend. I also misplaced my breadboard so I ordered one up with a bunch of passives. I can play with the resistor values a bit too.

dryja123, 2022-04-15:
@darthcloud flashed up the BT off firmware and the random button presses are gone.  I was unable to revert back to factory firmware by following the factory reset procedure.  That's ok, I'll get a programmer and program the device over USB.  I'm still going to tinker around with the dev kit this weekend too.

darthcloud, 2022-04-15:
ha crap sorry, I forgot the button task is started by the BT task :S. I'll make a couple FW tonight with different TX power setting.

dryja123, 2022-04-15:
Awesome, I can test it out on the dev kit. I’ll start with the latest stable firmware to duplicate the issue. I’ll flash up the test ones you send.

darthcloud, 2022-04-16:
Here builds for 3dbm, 0dbm, -3dbm, -6dbm, -9dbm and -12dbm
[v1.5.1_psx_bt_pwr.zip](https://github.com/darthcloud/BlueRetro/files/8498974/v1.5.1_psx_bt_pwr.zip)

3dbm is what was used in previous FW.

dryja123, 2022-04-16:
I was able to flash up to -6dbm before I wasn’t able to the adapter anymore. It wasn’t even coming up when I was scanning for Bluetooth devices with my phone, with the phone next to the adapter.

darthcloud, 2022-04-16:
You factory reset with those build. So did 0dbm or -3dbm had random input?

dryja123, 2022-04-16:
No, didn’t factory reset with those builds. I flashed up the firmware, rebooted the consoles, and tested. I had random inputs on everything I tested. Stock 3dbm, 0dbm, -3dbm, -6dbm. Wasn’t able to test further.

darthcloud, 2022-04-16:
if you disconnect 8v or 3.3v from PS and instead power externaly via usb?

Does that work?

dryja123, 2022-04-16:
I’ll give that a shot and report back.

darthcloud, 2022-04-17:
I made some experiment this weekend. In this post I'm looking mainly at the 8V supply. I think it's the proper way forward as it allow the ESP32 to use it's own 3V3 regulator and it's fused to a minimum of 0.8A (1A in practice) vs the 3V5 which is a bit higher fused to 0.6A.

I think the issue might be related to the ripple on the 8V line which I measured to be around ~700mV. The 3V5 got ripple too but it's much lower which is why I think we had some success with some system "fixing" the issue by switching supply source.
[image]

The ripple propagate through the LDO into the ESP 3V3 supply:
[image]

Looking at the X-Station ESP32 LDO, I see the supply 8V input is much cleaner:
[image]

Which in turn make the 3V3 clean too:
[image]

Looking at the PS schematic we can see the 8V supply follow 2 different path. The controller one which is EXT_+8V and the CTRL port use the common DIG_GND. I think the X-Station use the MOT_+8V and MOT_GND. That one goes though an extra LC filter (1uH + 47uF) (Resonant freq circuit?) and one final 4.7uF cap. The GND is (base on schematic) connected to a single point with digital GND.
[image]

So i've built the same circuit and added it to the UnoRetro by lifting the LDO input off the pad. The resulting ripple in the 8V supply is much lower but I didn't much improvement in the LDO 3V3 output. We are far away from what the x-station supply look like. 
[image]
[image]

Finaly holding unoretro reset make 8V ripple kind of go away:
[image]

But using a low power BT FW or even using the BT off FW have no effect on the ripple.

So no conclusion, maybe you might want to try to had the 1uH inductor and the 47uF and 4.7uF cap before the LDO input on the original 8V unoretro.

I think I'll try next weekend to use a capacitance multiplier:
https://electrical.codidact.com/posts/279580

dryja123, 2022-04-17:
This is a fantastic work and a great amount of information. I have the original kit, the kit with the 3.3v power, and a ESP32 dev kit. I’m a paint by numbers kind of guy and can hook what ever you suggest up to a breadboard and test; just let me know what I need to get. 

I’ll still test powering the unit off of external power.

darthcloud, 2022-04-19:
8bitmods guy say to try:
- CLK: 100 Ohm
- ACK: 22 Ohm
- DATA: 22 Ohm
- CMD: 220 Ohm
- ATT: 220 Ohm

dryja123, 2022-04-19:
Awesome, I have all of those except for the 22ohm resistors, I have 20. I’m sure the extra 2ohms won’t make a difference but if it doesn’t work I’ll buy some.

[97 more replies]

closed 2023-12-09 by darthcloud