darthcloud/BlueRetro #103: PS2 not recognizing analog inputs

Retron101, 2021-12-02:
I have tried use my switch pro controller for my ps2 but for some reason It won't recognize any analog inputs. My wiring is correct and I also noticed on the wiring diagram you have a connection to pin 8 on the db 25 connector even though pin 8 is not connected on the esp32 diagram.

darthcloud, 2021-12-02:
Which PS2 game?
Is it a game that do not activate analog by default?
Did you press the Analog button to see if it enable it? (Map to home button on switch controller by default)
Is it an official switch pro ctrl from Nintendo?

Retron101, 2021-12-02:
It's an official pro controller, the game I'm using is agent under fire and the home button does nothing.

Retron101, 2021-12-02:
I fixed it, it was a setting I had to change through the web browser.

closed 2021-12-02 by Retron101

darthcloud, 2021-12-02:
Could you had more details? Which settings was wrong?

Retron101, 2021-12-02:
I messed up the mappings when I was looking at the advance config tab. On mobile it's almost impossible because of how everything is squished, i made it so chrome shows the webpage as if I was using a desktop and now I'm able to see the settings as they should.