darthcloud/BlueRetro #104: N64 rumble does not working on Cruis'n world

raed143, 2021-12-02:
Been using NSO N64 controller with an N64 and it works great. Although, noticed that rumble does not work in Cruis'n world. Is there any setting I need to change to have it working?

Thanks.

darthcloud, 2021-12-02:
Did you set rumble as accessory in the web config?

If so did it work with other games?

raed143, 2021-12-02:
Yes, as I mentioned its working perfectly on, Starfox 64, Cruis'n exotica, Mario 64, Mario Tennis and most other games I tried. only exception is Cruis'n world. 

Thank you so much for creating and sharing this.

raed143, 2021-12-04:
Update: I noted that rumble sometime start and stops in less then a minute. Also this only effects player 1. There is no issue with player 2. When playing 2 players. Same issues happens with player 1 but with player 2 rumble works without issues. I used several controllers all have the same issues with player 1 and no issues with player 2.

darthcloud, 2021-12-19:
I had it working sometimes on Cruis'n World but sometimes it stoped working.

I made some change and now it look to always work. Can you try this build and let me know if it work for you
[1.2.1_wip2.zip](https://github.com/darthcloud/BlueRetro/files/7741699/1.2.1_wip2.zip)
?

raed143, 2021-12-20:
I have updated current ESP32 and also flashed new one with this firmware. And results are still the same. 

The game starts with rumble working and this its either stop and continues to continues to vibrate without stopping.  This is also not limited to cruisn world but its happens faster here. in most cases within few seconds from the start of the course. 

However, with player 2. there are no issues. it works fine and doesn't stop tested this for hours and its ok (same with ver 1.1.1, 1.2 and 1.2.1wip2).

Ive used only Switch NSO N64 controllers to test it. 

Its really weird that this happens only player 1.

darthcloud, 2021-12-20:
Ok weird, I played from Hawaii to the moon with no issues with Switch N64 ctrl aswell.

Sound more like a Bluetooth packet drop issue now.

darthcloud, 2022-01-26:
I tested this again in a multiplayer games rather than single player in case in made a difference. Still work perfectly for me.

Test again in v1.2.2 just in case, make sure rumble is set in advance web config for P1.

Once in game if rumble try pressing home button to toggle between mem pak and rumble to see if start working after. Press again if it didn't.

raed143, 2022-01-27:
Thanks for the update, tested the new update with both N64 NSO and switch pro controller and I am getting the same results. I am using original ESP32 and third party ESP32.

Using the following parameters:

Bootloader 0x1000
Partition 0x8000
N64spiff 0x10000

com 3 (shows up with I connect the board)

BAUD 921600

I verify that software updated correctly from the update section where it display current version which is 1.22

All other games works fine except for few including cruisin world which I am always using it for testing as its faster to start. 

The best thing about this mode is that it can connect 4 different controllers at a time and unlike other adopters where all register as controller 1 and you need to find which is connected to which port. this correctly identify the controller and not just that its easy to pick any controller rather then labeling each.  

And best of all you made this available for everyone to use. 

Thank you so much for all this efforts and I really hope that all this get finalized with with the final version we get compatibility with all games. 

Once done I am hopping to mode all my consoles with this. also its very easy to reconnect controllers as just press sync on start so I can use same controllers for consoles even though each will have different ESP board.

Regards

darthcloud, 2022-05-26:
Re reading this issue symptom I would bet you have a N64 digital which would explain the rumble issue with Player 1 only.

Latest firmware should fix your issue.

closed 2022-05-26 by darthcloud

raed143, 2022-05-26:
Yes, it works for some games perfectly without issues at all. But only for
one game rumble stops after few seconds. Tried that on multiple console but
yes all with n64 digital.

In this case maybe I'll remove  the n64digital controller connection or
maybe add a switch. I hope that this will resolve it. Thanks

darthcloud, 2022-05-26:
Just update to latest BlueRetro FW 1.6 its fixed