darthcloud/BlueRetro #108: Hard limit of number of paired controllers?

marceloMedeiros, 2021-12-07:
Hello. Ive been testing a blueretro device with many controllers and cables/systems and its working very nicely, really like it. 

One thing im having a hard time figuring out though is: is there a hard limit of number of paired controllers? i have several controllers and im able to pair many controllers, but im not sure if im doing somethin wrong but it seems that if i pair too many controllers, then some of the previously sucessfuly paired controllers start to lose its pairing, so i have to pair those again (wich in consequence seems to make other controllers to lose pairing).

Ideally i wanted to keep a vast number of controllers with many layouts paired and ready to use, since i would really love to use the blueretro as my primary way of plugging controllers on all my retro systems, but im not sure if theres a limitation there. 

Thanks for your time and for creating this awesome thing!

darthcloud, 2021-12-07:
The limit is for classic BT (BR/EDR) (and globally) is 7 controllers. BLE is limited to 3 only so far.

marceloMedeiros, 2021-12-07:
> The limit is for classic BT (BR/EDR) (and globally) is 7 controllers. BLE is limited to 3 only so far.
> 
> Not every mix of controller work well, so some controller will suffer in such setup. (like WiiU Pro)
> 
> PS4/PS5 look to be the best to use in such scenario.

I see...thats explains what im expriencing here! 

Even though 7 is plenty, i guess. Luckly im mainly using 4 dualshock controllers, so i should be fine. Im thinking of a setup similar to this:

- 4 dualshock controllers
- 1 SO N64 (if i can get hold of one eventually)
- 1 8Bitdo M30 (or 1 SO genesis 6 button, still debating wich one would be better)
- 1 SO Snes

If i can get that to work, i think it would cover most of my usage scenarios =)

darthcloud, 2021-12-07:
My answers was for simultaneous active connections btw. If you means pairing as in saving the connection keys you can save 16 connection keys for bt classic + another 16 for BLE.

The algo for overwriting keys once you reach the maximum probably suck, like it might overwrite the latest rather than oldest.

marceloMedeiros, 2021-12-07:
> My answers was for simultaneous active connections btw. If you means pairing as in saving the connection keys you can save 16 connection keys for bt classic + another 16 for BLE.
>
> The algo for [...]

Oh, i see!  i misundertood your previous message,thats even better! And yes, i was reffering to maximum connection keys that can be stored at any time. 16 is plenty! i think i have 12 or 13 controllers in total , which is more than i would realistically need. 

So i assume probably did something wrong on my end and caused some controllers to lose the pairing. I think ill just factory reset the blueretro and pair everything up again, and i guess it should probably work fine. If not, i can report back with more info.

One thing i might add as a suggestion is: including somewhere in the documentation the total number of connection keys blueretro can hold at the same time (16 connection keys for bt classic + another 16 for BLE). I looked it up but couldnt find that info in the docs, hence the creation of this issue here on GH. 

Thank you!

darthcloud, 2021-12-08:
I added a line in the wiki under https://github.com/darthcloud/BlueRetro/wiki/Home/_edit#pairing-bluetooth-controller

Thanks for pointing this out :)

closed 2021-12-08 by darthcloud