darthcloud/BlueRetro #116: [SUGGESTION] Wii support

almircanella, 2021-12-14:
Besides ogxbox, wii is among the best retro emulators for it's 240p support, and other features.

I love to use wi to play 240p consoles, and one of them is the genesis. I have the psx only blueretro device (will consoles modification). Updated to this day firmware.

- I have the 8bit do bluetooth m30 controller, on the ps2, it became unstable, the buttons did not work properly.
-  the m30 does not work with wii, and i tought it would work with some of the wii adapters (ps2 to gamecube, or ps2 to wiimote), but no dice either, it not even lights up in these adapters. 

I want to know if its a power supply thing, or software thing, that blueretro does not work with another controller adapters like the ones I said.

Thank you very much if you read till here.

darthcloud, 2021-12-14:
For Wii why not simply connect it via gamecube port? Most emu should support that.

And don't daisy chain adapters, this will kill latency for sure. Also all adapter cheat a bit in their implementation of protocol. Original console wont care, but plug two device cheating and they cant talk to each other.

You are trying to use a PSX blueretro on wii via adapter. (not a good idea) What you need to do is build a blueretro for gamecube, its easy only 3 wire to solder on an ESP32.

closed 2021-12-14 by darthcloud

almircanella, 2021-12-16:
Thank you for your time.

I know zero of soldering and have negative skills for understanding eletronic schematics, it's not for me. 

But thank you anyway. I will save money to buy pre-mounted from someone, a universal blueretro, and then, the cables for gamecube like you said.