darthcloud/BlueRetro #123: D-Pad PS4 freezes in Sega Rally 2 DC

scp-electrocution, 2021-12-26:
Hi, I installed an external esp32 card to my Dreamcast with ODE (MODE) and I noticed that in sega relly 2 the PS4 controller d-pad freezes for a few seconds then resumes working and continues so cyclically (seems to freeze after using the analog stick).
The problem also occurs with the WII U pro controller
I attach LOG:
[dreamcast sega rally 2 ps4 contr d pad issues.txt](https://github.com/darthcloud/BlueRetro/files/7776691/dreamcast.sega.rally.2.ps4.contr.d.pad.issues.txt)

darthcloud, 2021-12-28:
Hold on this for a bit, I'm getting a GDemu DC soon and will look into it then.

scp-electrocution, 2021-12-28:
The random d-pad blocking issue comes up with all controllers I've been able to try, xbox one, ps4 and wiiu.  I await news

darthcloud, 2022-01-05:
I played about an hour with both original DC wired and BlueRetro+PS4 and I didn't notice any

darthcloud, 2022-01-05:
I played about an hour with both original DC wired and BlueRetro+PS4 and I didn't notice any freeze.

The joystick feel different for sure but I think It mostly due to the DC one beings very smooth mechanically.

When you say freeze, is it something obvious to notice like control lag? Or is it very subtle?

What is the frequency the freeze happen? Continuously when using joystick? Every sec or two? Every 10 sec? Is the game playable still? Like you can get to next checkpoint?

How your Blueretro setup? All four port plugged in? Only one? Are you using Dreamcast specific FW or the universal one?

scp-electrocution, 2022-01-05:
The d-pad appears to freeze (stop working) after using the analog stick
The analog stick always works well, the d-pa starts working after 30 seconds, when you switch to the analog stick the d-pad stops working again, in loop, in the game and in the menus.
I use the specific dreamcast fw, connected externally with two cables (player one and two
[image]
)

scp-electrocution, 2022-01-05:
By pressing the d'pad and the analog stick together, the d'pad resumes operation

https://user-images.githubusercontent.com/91375683/148229719-2b416aa1-708d-4e99-a36a-7cabb09599f7.mp4

darthcloud, 2022-01-05:
Ok I understand your issue now, and I reproduced same behavior.

Very weird will look into it! Thanks for video!!

darthcloud, 2022-01-25:
I figured what was happening here. It's not really a BlueRetro bug per say.

The game prioritize analog joystick over the dpad input. If any joysticks value other than 0 is sent then the dpad value is ignored.

To minimum joystick value to make the menu move look to be around 15%.

So if you are over 0 and below 15% the joystick dont move the menu but you get in a condition were the dpad is still ignored.

I guess DC joysticks somehow always release to a neutral value of 0. Modern controller dont do so and rarely goes back to exactly 0.

BlueRetro got a dead zone to help with that but its often still not enough to fully get a 0 value. But the remaining values is low enough for games to ignore it 99% of time.

So I won't do anything to fix it, make sure to not touch joysticks at power on so that Blueretro init the center properly.

For this games I feel a easy wor around is to either use only dpad or only joysticks.

I bet that a really beat up original DC controller most likely got the same issue.

closed 2022-01-25 by darthcloud