darthcloud/BlueRetro #124: Xbox one view button no working like select on ps1/ps2

BrigiDios, 2021-12-27:
I have a PSUnoRetro with the latest firmware, everything works fine, but the view button doesn't work like the select button, it just doesn't do anything. My button is fine, I tested it on pc / xbox so I don't know what happened, I tried another controller (8bitdo) and this one doesn't have the problem. Do you know what is the problem?

darthcloud, 2021-12-27:
To fixup the mapping you need to update your controller FW with the xbox accessories win 10 app. This might fix other things as well.

https://www.microsoft.com/en-us/p/xbox-accessories/9nblggh30xj3

BrigiDios, 2021-12-27:
I updated to 4.8.1923.0 last firmware for xbox one controller, not difference, the view button still not working

darthcloud, 2021-12-28:
[firmware.zip](https://github.com/darthcloud/BlueRetro/files/7782185/firmware.zip)
Try this FW should fix it, look like it's broken since v1.0 and I somehow completely missed that!

BrigiDios, 2021-12-28:
Unfortunately, still the same ): updated to these firmware and not difference

darthcloud, 2021-12-28:
Power cycle everything and connect to OTA page and copy the webpage log at bottom.

BrigiDios, 2021-12-28:
I think you mean this

Requesting Bluetooth Device...
Connecting to GATT Server...
Getting BlueRetro Service...
Get Api version CHRC...
Reading App version...
App version: v1.0-beta psx_ps2_spiff
Init Cfg DOM...

Attach an image may be help

[image]

darthcloud, 2021-12-28:
Ok this means you are not running the FW i provided, can you try update again, when it reaches 100% wait a minute or so. Then power cycle again connect to OTA page, app version should be v1.2.

You selected the file BlueRetro_psx_ps2_spiffs.bin in the zip right?

BrigiDios, 2021-12-28:
For some weird reason it had not been updated correctly, but now it's already in the correct one that you provided me, but now the select and start button are the LB and RB bumper's respectively

Attach a image and log

Requesting Bluetooth Device...
Connecting to GATT Server...
Getting BlueRetro Service...
Get Api version CHRC...
Reading App version...
App version: v1.2-5-g3660bc5 psx_ps2
Init Cfg DOM...

[image]

BrigiDios, 2021-12-28:
I forgot to mention the view button and start button not working anymore

darthcloud, 2021-12-28:
Can you take screenshots of you advance config page after connect, the mapping section in particular:
https://blueretro.io/blueretro.html

Is it an official Microsoft controller or another brand?

BrigiDios, 2021-12-28:
Yes, it's a official controller from Microsoft, came with my xbox one s, also with this firmware that you provided me, the RB which is now start has some input lag

here the screenshots

[image]

[image]

[image]

[image]

[image]

[image]

[image]

BrigiDios, 2021-12-28:
about input lag, i think has my mistake, it was my program that had the error

darthcloud, 2021-12-28:
Which game are you using?, everything fine for me with that FW in Ape Escape on xstation.

BrigiDios, 2021-12-28:
I tried in ps1 with Tot Story 2, Resident Evil 3
These games are disc

In ps2 with Crash bandicoot 4, this game are in a HDD

Also I notice the X button from Xbox one controller is now triangle and Y button is now L1

I tried with 8 bitdo controller and everything is fine with this

darthcloud, 2021-12-30:
Ok my controller FW was recent but now latest after updating to 4.8.1923.0 I had same issue as you.

Try this FW:
[v1.2.1_wip5.zip](https://github.com/darthcloud/BlueRetro/files/7793521/v1.2.1_wip5.zip)

BrigiDios, 2021-12-30:
It works great! all buttons work again, thanks for the support

BrigiDios, 2022-01-03:
Hi it's me again, i was playing fine with my xbox one controller but the battery drains so while charging i plugged my 8 bit controller to keep playing but the inputs are wrong the select button is on various buttons , it is basically broken, I assume it conflicted with the new firmware

darthcloud, 2022-01-03:
Ok sorry about that, this one should be good for both:
[v1.2.1_wip6.zip](https://github.com/darthcloud/BlueRetro/files/7804261/v1.2.1_wip6.zip)

onlinewithryan, 2022-01-06:
Hi @darthcloud , I believe I have the same issue here and https://github.com/darthcloud/BlueRetro/issues/75

I am not sure if updating the firmware of my controller will fix this issue, but while debugging I found some places in the code that are causing the problem on my controller's current firmware.

I uncovered that BlueRetro is assigning the subtype of by device as an BT_XBOX_XS. When I hardcoded the subtype as BT_SUBTYPE_DEFAULT and flashed that firmware everything worked perfectly. 

The bug seems to exist in the fall through logic here

https://github.com/darthcloud/BlueRetro/blob/master/main/bluetooth/hci.c#L1284-L1293
```
void bt_hci_set_type_flags_from_name(struct bt_dev *device, const uint8_t* name) {
    for (uint32_t i = 0; i < sizeof(bt_name_type)/sizeof(*bt_name_type); i++) {
        if (memcmp(name, bt_name_type[i].name, strlen(bt_name_type[i].name)) == 0) {
            struct bt_data *bt_data = &bt_adapter.data[device->id];
            device->type = bt_name_type[i].type;
            device->subtype = bt_name_type[i].subtype;
            bt_data->flags = bt_name_type[i].hid_flags;
        }
    }
}
```
In the mapping there are two values for Xbox Wireless Controller, and the second is always used.  

```
    {"Xbox Wireless Controller", BT_XBOX, BT_SUBTYPE_DEFAULT, 0},
    {"Xbox Adaptive Controller", BT_XBOX, BT_XBOX_ADAPTIVE, 0},
    {"Xbox Wireless Contr", BT_XBOX, BT_XBOX_XS, 0},
```

I was looking into how to properly fix this since both controllers have the same Display Name. Looks like we can use SDP to get the product_id and set the subtype that way. I have an issue on my end where the device is initialized with the wrong subtype before I can update it. 

I then found these two issues and saw you had been working on firmware updates so I wonder if you had come up with a solution.

darthcloud, 2022-01-06:
Yes I fixed the name lookup function, you can try wip FW in this thread. But you will need update ctrl to latest FW with that WIP (src not published yet) MS changed from using Xinput buttons field to "dinput" style in latest FW. Not in WIP yet but I will fetch hid descriptor for xbox and try detect if it xinput so old FW still work.

Im not going to bother with vendor/product id as this dont really help anyway in the long run, controller that are different technically sometimes share same identical IDs, and some ctrl that are identical got different IDs. Often in linux things doesn't work just cause an IDs is missing from list. Ideally only using HID descriptor would be the trick.

So identification is best efforts base on name only and if no name found It use hid descriptor.

BrigiDios, 2022-01-06:
> Ok sorry about that, this one should be good for both:
> [v1.2.1_wip6.zip](https://github.com/darthcloud/BlueRetro/files/7804261/v1.2.1_wip6.zip)
> 

Sorry for late answer, but it works flawless, thanks for the firmware!

onlinewithryan, 2022-01-06:
> Yes I fixed the name lookup function, you can try wip FW in this thread. But you will need update ctrl to latest FW with that WIP (src not published yet) MS changed from using Xinput buttons field [...]

Not a blocker for me I can wait for the newest firmware src, I was just going a bit crazy reading through all of the different BT packet types. Now that I'm more comfortable in the code I have a handful of ideas I want to implement as well as contribute to making the webapp more visually stunning.

closed 2022-01-07 by darthcloud

darthcloud, 2022-01-07:
Maintaining backward compatibility turnout to be a complex rabbit hole. So for now using latest MS firmware for Xbox One S controller is required with v1.2.1.

commit 8bf14d4d (2022-01-07): [XBOX] Fix controller mapping for XBOX FW 4.8.1923.0

Fixes #124