darthcloud/BlueRetro #131: Rapoo BLE mouses stuck at init state

darthcloud, 2021-12-30:
The HID attribute range is bigger than what is actualy used and this make the find info state get stuck.

darthcloud, 2021-12-30:
Try this firmware with a fix:
[v1.2.1_wip5.zip](https://github.com/darthcloud/BlueRetro/files/7793547/v1.2.1_wip5.zip)

Betonar, 2021-12-30:
Updated and tested with Rapoo 7200M. I can confirm mouse work as expected. Pairs instantly and is able to reconnect after I turn it off/on. Tested few only mouse games, no issues found. Also tryied with DS3, no issues what so ever. So far I dont know how to connect more devices at the same time so cant provide more feedback. If something pops up while I play in single device games I will let you know

commit 81d2a194 (2022-01-07): [BT] Handle oversized range in find info state

Fixes #131
closed 2022-01-07 by darthcloud