darthcloud/BlueRetro #145: Some Atari Jaguar games are unplayable due to button mapping issues

DoomyDoomer, 2022-01-16:
I have tested all 50 retail cartridge games as well as 4 CD games. The following games have a button mapping issue:

- **Atari Karts** 
On PS4/5 controller Pause = A, L1 = D-Pad R, D-Pad L = 4, R Thumb = D-Pad L. D-pad R, U, D and other buttons not working
- **Breakout 2000**
Same as Atari Karts
- **Missile Command 3D**
Same as Atari Karts
- **Myst**
L1 acts like D-Pad Left and R3 acts like D-Right 

My test was with a PS4 and PS5 controller. I've tried to remap these games in the advanced web config but it didn't seem to have any effect.

darthcloud, 2022-01-26:
Breakout 2000 issue look different than Atari Kart & Missile Command.

If you simply press right and B to start the game right away everything look fine to me.
If you press down in the menu, then the game goes in an infinite loop.

[v1.2.3_wip1.zip](https://github.com/darthcloud/BlueRetro/files/7945373/v1.2.3_wip1.zip)

Try this build, it will fix the infinite loop in breakout 2000, other than that, this game look completely fine to me.

Will look into the other next.

darthcloud, 2022-02-02:
I can't really fix the other 3 games, the games is simply reading back the JOYSTICK/JOYPAD register too quickly after writing to it. TTL chip can handle it since they probably transition output in the 10ns, Using pretty much all the trick I could think off the ESP32 can only do between 300ns and 400ns transition.

[image]
[image]

This make those game read the previous buttons bank set. This look quite stable.

As a workaround, I a made a preset called "Jaguar cycle shift" (see https://blueretro.io/blueretro_presets.html) and using that both myst and atari kart are playable properly. You can't use that in the working games however. To help navigate the GameDrive I mapped the non-shifted d-pad on the left joystick.

Missile Command is still a bit broken, the main menu up/down dont work at all. The timing of bank transition is quite brutal at 1us (vs 8us for Atari Kart & Myst), but once in gameplay it kind of work?

[image]
 
An alternative workaround for GameDrive people would be to make rom hack of the games with some NOPs after setting JOYSTICK register.