darthcloud/BlueRetro #146: Switch controller (All type) multiplayer 2nd+ heavy latencycy

darthcloud, 2022-01-17:
Switch controller somehow don't play nice with eachother when used simultaneously. Mixing one with another type like PS or Xbox work fine.

When 2 or more SW are used, often the first one will be absolutely fine but the other will often have extreme latency.

Another behavior is that sometimes pressing buttons on first ctrl right after 2nd ctrl connection cause the 2nd to get in a weird state where led indicator is reset to one and no report is received.

Workaround: only use one SW controller, use other ctrl type for multiplayer.

sskylar, 2022-03-10:
Confirming the same behaviors with the following Switch-based controllers:

- PowerA Wireless GameCube Style Controller
- HORI Wireless Battle Pad (NSW-274U)
- 8bitdo SN30 Pro+
- 8bitdo N30

novamax89, 2022-08-01:
Hello. I found out that if I pair an Xbox series x controller first, then I pair a switch controller they both works, but if I turn off the Xbox controller and pair the switch controller now both switch controller works perfectly. This is a nice workaround to let the controller always works without making many tries, but also means that there is maybe a software problem and not an hardware one if they can both work fine with this method

novamax89, 2022-08-02:
Do you think you can solve the problem updating the firmware? @darthcloud

raed143, 2022-10-29:
Is this issue is still open? or shall I install multiple boards? (1 for each 2 controllers).  If I did. will it be possible to set each board to only accept to controllers and the other board assign first controllers connected to 3 and 4 instead of starting with 1 and 2? Thanks

wtfbbqhax, 2022-12-16:
I came here to report the same issue with regular switch pros and NSO N64 controllers. 

In my case, I can not get the subsequent controllers to work at all (10+ tries)

Traxus7, 2023-01-09:
Reporting the same issue with multiple NSO N64 controllers. I have a single BlueRetro unit with 4 N64 cables and trying to pair with multiple NSO N64 controllers results in them stepping on each other. Pairing one works fine, but pairing a 2nd controller results in both controllers going in and out of connection. Completely unstable. The 2nd controller's #1 LED will go solid occasionally, like it's trying to take over channel #1.

I nabbed 4 NSO N64 controllers for this; if it worked properly this would be my dream N64 setup. Is this something that can be patched or am I better off shopping for 4 separate BlueRetro dongles?

raed143, 2023-01-09:
In my case 2 NSO controllers worked well on one board. So installing 2 boards with a switch to turn on/off the second board solved the problem.You can do 4 as well but in this case you better install switches to turn the additional boards on an off to prevent the controllers from pairing to other boards if you’re playing single or 2 player game.

wtfbbqhax, 2023-01-10:
I asked Darthcloud on Discord a few weeks ago about debugging this issue and apparently it is non-trivial. 

The only workaround is to have individual adapters for each controller.

darthcloud, 2023-01-10:
Its not a solution its a workaround at best, using 4 radio when you could use one doesn't make sense.

Traxus7, 2023-01-12:
Yeah I'm just an end-user but I assume since BlueRetro is capable of pairing 4 pads with a single board that would be ideal at least to minimize BT interference, as opposed to having 4 separate dongles each with their own transmitter. Shame it sounds like a tough nut to crack; four N64 NSO controllers paired with one cohesive BlueRetro unit on og N64 hardware would be so nice.

Regardless, I just want to say thanks for everything darthcloud! BlueRetro is absolutely amazing. Take your time. Health and family come first! I'll be looking forward to all the cool things the future brings.

anherooftime69, 2023-02-01:
Thank you for all of the work you put into this project @darthcloud.

I bought a pre-made core and connectors for each of my systems from Aliexpress and quickly learned of this issue on my own. It's nice to see that it's a known issue, and that you are not ignoring it. For that I am grateful!

And not that you are responsible at all for it, but I just wanted to point out that the vendor is not aware of this issue, and they imply that it works fine with multiple Pro Controllers.

raed143, 2023-02-01:
There is only one authorized vendor in Aliexpress called retroscaler. The others just copied the open source instructions to profit out of them.

anherooftime69, 2023-02-01:
@raed143 that's unfortunate, I bought them from [BlueRetro Gaming Store](https://blueretro.aliexpress.com/store/1102191985).

davidreeves, 2023-03-05:
I found this to still be a problem on firmware 1.8.3 for the gamecube dongle.

ravenrue, 2023-03-10:
Hi, was wondering if there was an update to the "6.3 - Dual Joycon Pairing Guide"? The directions are for an older version of the blueretro config. The screenshot doesn't match what we currently have. Thoughts? Am I wrong? Please help. I'm trying to sync my joycons to the gamecube. Thanks!

AugusDogus, 2023-03-25:
I've created a bounty for this issue @ [image]

wtfbbqhax, 2023-03-27:
Can I somehow contribute to your bounty?

TriVoxel, 2023-03-28:
@wtfbbqhax Just create a new one for this issue.

Traxus7, 2023-04-03:
omg you guys are heroes, thank you, would love to see this solved!

icd2k3, 2023-05-17:
Many thanks to darthcloud and others for looking into this...

In the meantime, does anyone have other controller recommendations for N64 and GC? I suppose I could also look into doing 1 esp32 internal board per port, but that sounds... tricky

I was hoping to use the RetroFighters NSO Brawler and PowerA's NSO GC controllers because the button layout and feel is close to the originals. Anything similar that has better multiplayer compatibility?

Either way, thanks to those looking into this and contributing bounties

raed143, 2023-05-17:
> Just to confirm, this isn't a problem with multiple dongles, right @darthcloud?
> Obviously an internal install would be preferred (and cheaper), but if multiple dongles allows me to use 4 NSO controllers without issue, then it's worth it for me (until the single board install can be worked out)

Yea, as a temporary solution I have installed multiple chips internally and they work fine. If you buy they esp32 chips separately they are cheap around USD 3 each and installation is very easy on the N64 (only 3 cables and 3.3 volts power so no need for additional converters. Sent from my iPhone

raed143, 2023-05-17:
> @raed143 do you happen to have an example pic of your install?
> I just kinda wanna get a visual idea of what's involved. I do have 4 of the tiny chips nostalgic-influences uses, so I should be able to stash the somewhere - the single chip in his example has a lot of connections for port detection...
>
> For port detection, each board will probably need 4 wires if I understand correctly: IO, 3v, Grd, IO2 (port detection)
> Sorry if this is too off topic, but it would be a sweet workaround for this issue if I can get it working

I don’t have pictures with me now but it’s very simple. Each controller has 3 pins, one for power 3.3v, the middle for data and the third is neutral. For the data pin just follow cable instruction listed here and the other 2 pins will be for power and neutral for each chip. U only need to use one per chip. If you will installed 2 chips that obviously all will be used but if you’re going to install 2 then use the one for p1 and p3. I installed 2 chips and is working fine. I noticed that once I add the third controller I start getting issues but some faced issues even with the second controller.This account have posted diagrams, but mainly for using HW2 edition and are made very clear but you can easily use the same wiring but ignore connection points at the board and just use the controller port pins.Save those GBA boards that have the faulty and irreplaceable power management ICs by giving the CPU & RAM a new home and a new life … 😌Nostalgic Indulgencestwitter.comI will also advise you to install, red and blue led instructions are also available here to know which controller is connected and which board is still seeking a controller.I have also added some switches using the vents on the bottom (without damaging the shell) to turn on and off each board individually.  I hope that this helps you. But n64 board is the easier. 1 cable per controller vs 15 cable on Neogeo not to mention that each cable has to go through 5v to 3.3v converted so it’s 60 solder point per controller (120 for both) while n64 all 4 controllers only requires only 12 points + 2 for each additional board. Sent from my iPhone

icd2k3, 2023-05-17:
Thanks @raed143 that does help - I only want active port detection (HW2) on the 1st port anyways, so maybe it wouldn't be too hard to wire it up that way and do the other 3 chips as HW1

icd2k3, 2023-06-14:
With 2 NSO SNES controllers I don't see any latency, but the 2nd controller always goes into the state where the indicator LED is reset and it loses connection. Sometimes it will work for up to 5min, but it always reverts to that state.

Bumping the bounty to $250 total 

[image]

icd2k3, 2023-06-14:
Does anyone know if this problem happens with the 8BitDo mod kits? For example the [original](https://shop.8bitdo.com/products/8bitdo-mod-kit-for-original-snes-sfc-controller-old-edition) or [classic](https://shop.8bitdo.com/products/8bitdo-mod-kit-for-snes-sfc-classic-edition-controller-new-edition) SNES controllers or [N64](https://shop.8bitdo.com/products/8bitdo-mod-kit-for-original-n64-controller) controller?

Or do those mods behave like switch controllers too?

Traxus7, 2023-06-16:
> Does anyone know if this problem happens with **the 8BitDo mod kits**? For example the [original](https://shop.8bitdo.com/products/8bitdo-mod-kit-for-original-snes-sfc-controller-old-edition) or [classic](https://shop.8bitdo.com/products/8bitdo-mod-kit-for-snes-sfc-classic-edition-controller-new-edition) SNES controllers or [N64](https://shop.8bitdo.com/products/8bitdo-mod-kit-for-original-n64-controller) controller?
> 
> Or do those mods behave like switch controllers too?
> 
> #### Edit
> Apparently they support xinput, so I'm gonna bite the bullet on 1 oem snes controller and 1 mod kit and test to see if it works

In BlueRetro 1.83 on a Retro Rosetta I can confirm the 8bitdo N64 modkit does NOT suffer from the Switch latency issue, either in Switch mode or Windows/Android mode. I've tested my four 8bitdo modkits paired simultaneously in Switch mode, and also one NSO N64 controller paired with three 8bitdo modkits in Switch mode. They all performed beautifully.

I also tried two NSO N64 controllers and immediately the latency of the second controller was obvious, unplayable.

However, the 8bitdo does suffer from multiple button mapping issues in both modes and rumble currently doesn't work. See the issue thread on it. I imagine it'll be an easier fix than the Switch controller latency once darthcloud gets a look at it but I have no clue.

pceslayer, 2023-06-22:
> Does anyone know if this problem happens with **the 8BitDo mod kits**? For example the [original](https://shop.8bitdo.com/products/8bitdo-mod-kit-for-original-snes-sfc-controller-old-edition) or [...]

It doesn't seem to impact the SNES or NES kits either. Ive played a bunch of bomberman

> In BlueRetro 1.83 on a Retro Rosetta I can confirm the 8bitdo N64 modkit does NOT suffer from the Switch latency issue, either in Switch mode or Windows/Android mode. I've tested my four 8bitdo [...]

I can confirm I the NES 8bitdo kits do NOT have the latency issues either. I have 4 NES Kit controllers that work perfectly. I wouldn't think the SNES kits would have the problem considering the NES kit seem to be identified identically in Blueretro. I only have two SNES kit controllers so I can't test multiple there, but the NES and SNES play nicely together with no issues.

jonah360, 2025-02-14:
> Switch controller somehow don't play nice with eachother when used simultaneously. Mixing one with another type like PS or Xbox work fine.
> 
> When 2 or more SW are used, often the first one will be absolutely fine but the other will often have extreme latency.
> 
> Another behavior is that sometimes pressing buttons on first ctrl right after 2nd ctrl connection cause the 2nd to get in a weird state where led indicator is reset to one and no report is received.
> 
> Workaround: only use one SW controller, use other ctrl type for multiplayer.
> 
> Workaround 2: Disconnect SW controller with Boot buttons on ESP and then reconnect. Do so until all controller are responsive. It usually take less than 5 

Has this been resolved in 2025 @darthcloud

darthcloud, 2025-02-14:
This issue is still open so no

Arial-Bold, 2025-04-16:
> This issue is still open so no

What work has been done on this? It's been a few years and we have heard nothing. This is a major flaw as many people will want to use the PowerA Gamecube Switch controllers and now the new Nintendo Switch2 Gamecube Classics controller.

TriVoxel, 2025-04-16:
@darthcloud Hey, check out these two issues:

1. [DanielOgorchock/linux](https://github.com/DanielOgorchock/linux/issues/33#issuecomment-2790843365)
2. [ValveSoftware/steam-for-linux](https://github.com/ValveSoftware/steam-for-linux/issues/7631#issuecomment-2800865186)

Apparently there was a workaround found where renaming the bluetooth device to “Nintendo” fixes the Switch controller vibration issue. Perhaps that would work here?

darthcloud, 2025-04-17:
@TriVoxel GOD if that turn out to really fix, then it's infuriatingly stupid.

I tired with 3 official Switch ctrl like 10 times with the adapter named Nintendo and it worked perfectly on each try. I rollback to BlueRetro name and it failed right away.

I'm not declaring victory yet, I want to test it around 100 time to be sure. Unfortunately I can't tonight.

Anyway try theses build witch name the adapter Nintendo and let me know:

[v25.04-beta-26-gd4e1c92a_hw2.zip](https://github.com/user-attachments/files/19786536/v25.04-beta-26-gd4e1c92a_hw2.zip)
[v25.04-beta-26-gd4e1c92a_hw1.zip](https://github.com/user-attachments/files/19786535/v25.04-beta-26-gd4e1c92a_hw1.zip)

Arial-Bold, 2025-04-17:
> [@TriVoxel](https://github.com/TriVoxel) GOD if that turn out to really fix, then it's infuriatingly stupid.
>
> I tired with 3 official Switch ctrl like 10 times with the adapter named Nintendo and it [...]

I can in fact say this does work, however there is an odd button mapping with this update. It looks like the a and b buttons are swapped. I have been able to correct this with a custom button mapping, but something to take a look at. Thank you so much for getting this fixed.

darthcloud, 2025-04-17:
> 
> I can in fact say this does work, however there is an odd button mapping with this update. It looks like the a and b buttons are swapped. I have been able to correct this with a custom button mapping, but something to take a look at. Thank you so much for getting this fixed.

Mapping look fine to me, if testing on a GameCube note that BlueRetro default mapping is positional rather than name based. So GC A is map on B for a Switch Pro ctrl. You can use minus button to toggle to name base mapping without needing to go into the mapping config.

darthcloud, 2025-04-18:
Well I got this backward, BlueRetro use the default report 0x3F, so maybe they use the native 0x30 one

TriVoxel, 2025-04-18:
Wow WTF? This is clearly malice on the part of Nintendo… they straight up break the Bluetooth connection if it’s not being played on their system… yikers

TriVoxel, 2025-04-18:
Anyone know if Windows has this issue at all? I know Apple TV does.

TriVoxel, 2025-04-18:
> I can in fact say this does work, however there is an odd button mapping with this update. It looks like the a and b buttons are swapped. I have been able to correct this with a custom button [...]

Well, rumble isn’t working but, 3 official Switch controllers simultaneously is!! Could you enable rumble BTW?! I’m assuming you disabled it for disconnection reasons as that causes disconnection on Linux. Mappings are all messed up but otherwise, pretty good!

Here’s a clip of me testing 3 controllers at once:

https://youtube.com/shorts/1YYuJyTFHuw

darthcloud, 2025-04-18:
Rumble work, make sure you enabled it in the config

darthcloud, 2025-04-18:
Mapping is fine too, BlueRetro default mapping is position base, not name base.

TriVoxel, 2025-04-18:
> Mapping is fine too, BlueRetro default mapping is position base, not name base.

Oh. Honestly, can we PLEASE get an option to change this to name-base? I've never cared that things are positionally the same. I get very frustrated when I press A and it's X or something... perhaps a separate issue but, one of the main reasons I don't use my BT adapter that often. I invariably need to go into the Blueretro config and tweak stuff because the letters are all mis-matched. IDEK how GC controllers don't map to GC controls tbh... I digress!

darthcloud, 2025-04-18:
If you read my earlier answer to someone else in this thread or read the doc you would know you can just press minus button to swap it.

https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-System-Specific-User-Manual#mapping-15

darthcloud, 2025-04-18:
PowerA GC controller can't be distinguished from PowerA "Pro" so that's why you need to pres minus to swap it yourself.

TriVoxel, 2025-04-18:
Glad to report, I just played like 5 hrs of Burnout 2 Championship mode on my Gamecube with an official Switch Pro Controller w/ rumble enabled. BTW, that game with Switch haptic rumbleSO GOOD! I’ve been missing out!

Also, played at least an hour with my Power A Switch GameCube controller. No problems. No input lag, no disconnects, no rage quitting trying to get the air time challenge in the tutorial…!

Unbelievable! This was such a terrific experience!

icd2k3, 2025-04-18:
This all sounds very promising... I have 4 Exlene GC controllers and LaserBear's internal mod. I'll try to test connecting all of them in switch mode this weekend. This would _almost_ be a perfect multiplayer GC setup for me (4 controllers, no lag, all rumble... only problem is no analog triggers, but I can live with that)

icd2k3, 2025-04-19:
Ah dang, the beta seems to work with 3 controllers in switch mode, but connecting a 4th causes all of them to disconnect and BlueRetro to reboot - [here's a video of the behavior, if it helps](https://www.dropbox.com/scl/fi/j9jywrenyogho4f9mdd0x/IMG_2646.MOV?rlkey=90waapwd9vywag3ibwxjuh2dc&st=ezywb064&dl=0) - lmk if I can provide anything else!

darthcloud, 2025-04-19:
Try latest beta https://github.com/darthcloud/BlueRetro/actions/runs/14544498684
I probably fixed that this morning

icd2k3, 2025-04-19:
> Try latest beta https://github.com/darthcloud/BlueRetro/actions/runs/14544498684 I probably fixed that this morning

That seems to have done it! 4 Exlene GC controllers paied in switch mode with no lag! trying to test rumble too but the advance config keeps giving "unresponsive" errors in Chrome

Other than that, this is awesome though, great work!

TriVoxel, 2025-04-19:
> This all sounds very promising... I have 4 Exlene GC controllers and LaserBear's internal mod. I'll try to test connecting all of them in switch mode this weekend. This would _almost_ be a perfect [...]

I had a similar issue with 3 Power-A Switch GameCube controllers paired on the previous beta. It happened two or three times total in ~3.5 hrs of gameplay so, not a big issue. I’ll have to get some friends together to test the new beta!

TriVoxel, 2025-04-21:
> Try latest beta https://github.com/darthcloud/BlueRetro/actions/runs/14544498684 I probably fixed that this morning
> 
> Always pair them first "alone", then disconnect them. Once all paired to BR, it's easier to just reconnect them. But yeah even this way reconnecting the 4th will take more time.

Hey, got the family back together and we played multiple titles, this time with 4 players using Switch controllers. I have the LaserBear internal GC mod (HW2).

Player 1 was an official Switch Pro controller, players 2-4 were Power-A GameCube Switch controllers. We had no issues at all whatsoever with disconnections using this new beta you linked!! We played for ~4 hrs with 4 controllers connected simultaneously most of the time. Rumble was enabled for all 4 ports, though the Power-A controller does not have a rumble motor.

darthcloud, 2025-04-27:
Fixed in v25.04.

closed 2025-04-27 by darthcloud

darthcloud, 2025-04-27:
@sskylar I'm not using that service, you should probably try to get a refund, you can support on github, kofi or Patreon. Thank you.

darthcloud, 2025-04-27:
Thanks @AugusDogus, really appreciate your support!

TriVoxel, 2025-05-05:
@darthcloud [This guy’s comment](https://github.com/DanielOgorchock/linux/issues/33#issuecomment-2849798054) explains the issue well in more detail.

It seems like the controller actually uses a completely different mode if the Bluetooth adapter name matches a few values. There’s 3 different modes. Naming it to “Nintendo” uses a higher “active” polling rate with the standard HID protocol.

There are two other modes which are the ones actually used by the Switch itself. Presumably, you would be getting the exact same signals as a real Switch if you used one of the other protocols. May be worth testing. Could have better latency or something.