darthcloud/BlueRetro #155: PS1 (single motor) rumble not working, at least on PS2 console via backward compatibility

patters-syno, 2022-01-28:
I use a PSUnoRetro device (from willsconsolemodifications.co.uk) running BlueRetro, with a PlayStation 2 console and a DualShock 4 controller. I have enabled BlueRetro rumble via the web config, and it works for PS2 games.

I have noticed that there's no rumble at all for Porsche Challenge PAL - a PS1 game working on the PS2's backwards compatibility. Using a wired DualShock 2 pad rumble does indeed work, so it might logically be expected to work with BlueRetro.

However it does feel as if only the right-hand vibration motor is used when using a real DualShock 2 pad. From memory of owning this game back in the day, and the PS1 Dual Analog controller, I believe this game was one of the first to support that type of controller.

In a somewhat strange decision by Sony, the Dual Analog controller had its rumble motor removed for the US and European markets before it launched, but it did have a single rumble motor in Japan (https://en.m.wikipedia.org/wiki/Dual_Analog_Controller). So it would make sense that this game supports only one vibration motor, even if the feature only worked with the Japanese Dual Analog pad, and later with the DualShock and DualShock 2 controllers, themselves enhancements of that original design which remain backwards compatible with it.

Supporting this theory, Porsche Challenge is described here as being a game which indeed included support for the single vibration motor of the Dual Analog controller: https://en.m.wikipedia.org/wiki/DualShock

So... is it the case that the current BlueRetro rumble implementation only activates a DualShock 4 controller's vibration motors when output for the left-hand DualShock 2 motor is requested, and perhaps ignores requests for the right-hand motor only? Or is there some other problem breaking vibration support for Porsche Challenge specifically?

I originally raised this as a query inside another issue ticket here:
https://github.com/darthcloud/BlueRetro/issues/20#issuecomment-976030915

I updated that I have seen the same issue in Gran Turismo PAL so it doesn't seem to be isolated to Porsche Challenge.

@darthcloud confirmed that it should be working, a single motor request should indeed trigger BlueRetro to rumble.

My PS2 is a model SCPH-90003 super slim (with built-in PSU), and I have updated to BlueRetro 1.2.2. This version remains affected by the issue.

mario-64, 2022-06-13:
BlueRetro's PS2 rumble support is incomplete. This is noted in issue 20. Hopefully it'll get fixed soon.

patters-match, 2023-08-12:
Any update on this? For a device targeting retro systems, complete PS2 support is surely a fairly important goal?

mario-64, 2023-08-12:
> Any update on this? For a device targeting retro systems, complete PS2 support is surely a fairly important goal?
Darthcloud hasn’t responded to any posts regarding this issue so don’t expect it to get fixed. I don’t use Blueretro for PS1 or PS2 because of it.

darthcloud, 2024-05-29:
Can someone test this beta, this add dual motor and variable force support for Playstation and Dreamcast

https://github.com/darthcloud/BlueRetro/actions/runs/9286878322

rpf16rj, 2024-09-17:
> Can someone test this beta, this add dual motor and variable force support for Playstation and Dreamcast
> 
> https://github.com/darthcloud/BlueRetro/actions/runs/9286878322

Hi, I didn't download the artifact, because is expired.

darthcloud, 2024-10-06:
Fixed in v24.10:
https://github.com/darthcloud/BlueRetro/releases/tag/v24.10

closed 2024-10-06 by darthcloud

patters-match, 2024-10-11:
I have finally found some time to update to v24.10 and test. Unfortunately the rumble still isn't working in Porsche Challenge (PSX) running on a PS2 in backward compatibility mode. Works fine with a regular DualShock2.