darthcloud/BlueRetro #156: Getting Blueretro to work on ESP32-PICO-DevKitM-2

bixxewoscht, 2022-01-31:
Debug Log
[PICO Log.txt](https://github.com/darthcloud/BlueRetro/files/7972944/PICO.Log.txt)
:

darthcloud, 2022-01-31:
Look like the Bare Metal hack for CPU1 make it crash.
Will look into it when I receive my devkit.

bixxewoscht, 2022-03-07:
Thank you so much for your effort.
I just flashed it onto my devkit and will try it out once I have some free time.

The obvious pros of the Pico are that it enables even smaller builds, but the main question will be if you think the additional programming work is worth it.
I'll report back once I have some results.

darthcloud, 2022-03-07:
No worries it's worth it for sure, I'll probably have official support in about a month.

bixxewoscht, 2022-03-13:
I had some time to test this on N64 with the PICO Devkit today. Everything worked as it should, also changing the config and activating rumble. Looks good so far.

darthcloud, 2022-03-13:
let me handle issue closing ;)

bixxewoscht, 2022-03-13:
Sorry, somehow GitHub did not let me comment without closing xD

commit ee6c0938 (2022-05-23): [SYSTEM] Detect PICO package and use GPIO20 for LED

Closes #156
closed 2022-05-23 by darthcloud