darthcloud/BlueRetro #168: PCE Controls

pmgducati, 2022-02-16:
Unlink the gamepad & GamePadAlt default mapping, so this way 2 btns mode will have I on right face, and II on bottom face, and select/III on left face.

https://github.com/darthcloud/BlueRetro/discussions/167

darthcloud, 2022-02-16:
btw thanks a lot :) @pmgducati, v1.3 probably out by end of February.

pmgducati, 2022-03-01:
Do you think this feature will be included in 1.4?

darthcloud, 2022-03-01:
v1.3.1

darthcloud, 2022-03-02:
See here my proposal for the new buttons mapping coming in v1.3.1

https://docs.google.com/spreadsheets/d/e/2PACX-1vT9rPK2__komCjELFpf0UYz0cMWwvhAXgAU7C9nnwtgEaivjsh0q0xeCEiZAMA-paMrneePV7IqdX48/pubhtml

Please review the PCE 2/3 btns & PCE 6 btns colums

pmgducati, 2022-03-02:
I think it would be beneficial to have the I and II in their native spots (II = B/PAD_RB_RIGHT and I = A/PAD_RB_DOWN) as 99% of the game will be 2 buttons.  With that said I think III/Select could be either X/PAD_RB_LEFT or Y/PAD_RB_UP as users that are used to the original controllers would reach for select anyway, this is a new button that doesn't have a muscle memory for players.

darthcloud, 2022-03-02:
If you check NES mapping B is RB_LEFT and A is RB down.

This is to match how SNES Mario world play vs the NES SMB, also technically when using a SNES pad on a NES Y is B and B is A.
To me this is more natural.

I know past that point in time the other way around became the prominent default but I really hate that personally.

So I'm not changing that, cause otherwise if you use 2 btns BT controller like Wiimote or Switch NES you end up missing a buttons.

pmgducati, 2022-03-02:
Looking at the NES and the Wiimote in the I would still like to chat about the idea of using RB_RIGHT and RB_DOWN for PCE and NES.  This would make the Wiimote easier to play (I circled the current setup in Red and proposed in yellow).  This would also give the physical  layout to match the "Horizontal" buttons on the NES and TG-16/PCE controllers and Ladder back to the SNES B/A Layout. Let me know what you think.

[image]

darthcloud, 2022-03-02:
Actualy Wiimote mapping is incorrect., right now, but it going to be change to 1 = L and 2 = D

I understand you point but ultimately my preference win ;) , I just prefer 2 btns games to use L and D on 4 btns ctrls.

pmgducati, 2022-03-02:
Hey, I tried :) at least they will be on the face!  With all that said lets proceed!

darthcloud, 2022-03-08:
Fixed in [v1.4](https://github.com/darthcloud/BlueRetro/releases/tag/v1.4)

closed 2022-03-08 by darthcloud

pmgducati, 2022-03-13:
Hey, Just got back from a vacation to this wonderful post!  I tested 1.4 and everything is working perfectly!  Thanks for everything!