darthcloud/BlueRetro #201: Random disconnections on ps2 and Gamecube

MinZe25, 2022-03-04:
Hello, awesome project!
It seems that sometimes the esp32 kinda stop working.
I've seen it with Ps2 and GC both.
With ps2 it stops receiving inputs and continues sending the current input non-stop. connected with original DS4s. When connecting two DS4s to one esp32 with two Ps2 cables, this happens to one of the controllers, not both, if given enough time it will happen to both, just not at the same time.
With GC it just disconnects the esp32 entirely.
Those things have happened with different esp32s, but all of them were the same chip.

darthcloud, 2022-03-04:
Are you sure your DS4 are fully charged?

Are you using official ESP32 devkitc from espressif bought from either digikey or mouser?

Can you get some pictures of your install were I can see connection easily.

If possible, disconnect the ESP from console 5v supply and power from USB. Try to get log when DS4 disconnect:
https://github.com/darthcloud/BlueRetro/wiki/Getting-BlueRetro-debug-logs-via-Serial-port-Windows-10

MinZe25, 2022-03-06:
Ds4 are fully charged.
I'm using a custom pcb that's basically a normal esp32 but instead of pins it has terminal blocks.
When it stops working nothing responds, I cannot pair another DS4

kingknuts, 2022-03-19:
Happens the same to me also with a Xbox one series X controller. Sometimes the controller keeps connected, but it don’t send any button presses. How can I get the logs??

kingknuts, 2022-03-19:
This happened right now.
[image]
(Please plug again an analog controller (DUALSHOCK) or an analog (DUALSHOCK2) in controller port 1 to continue.

darthcloud, 2022-03-19:
Try to power adapter fron ps 3.3v to esp 3.3v, instead of using 8v on 5v input.

Which game you use?

kingknuts, 2022-03-19:
All right, but I have them welded, I will try it tomorrow. I’m not very good at welding.
The games in the picture that throws that message is:
Grand Theft Auto, Liberty City Stories.

I observed similar behaviour in Gran Turismo 3 A-Spec (controller connected, but no steering and full throttle)
AirBlade (Controller disconnected screen)

kingknuts, 2022-03-20:
Using the 3.3 none of my controllers pair. Looks like there’s not enough power to turn on Bluetooth. The computer cannot connect also. I tried on a fat PS1 and PS2 slim, and got the same results

darthcloud, 2022-03-20:
Make sure your 3.3 is on the ESP 3.3v and not the 5V ones.

But its possible, this is why i was using 8V myself but then many people reported random input using 8V and none with 3.3V.

I guess PS power need more investigation!

kingknuts, 2022-03-20:
Yes, it was properly powered ;)

kingknuts, 2022-03-20:
Ok, if you need to try anything else, please let me know.

KChito-RGB, 2022-07-06:
I'm having the same issue with random disconnections on Game Cube using an Xbox Series X controller.

KChito-RGB, 2022-07-19:
Today I updated to the new version and my sync problem has disappeared :D

darthcloud, 2024-04-15:
This is kind of old.

I think v1.9.2 stabilized things a lot.

If any of you still got random disconnect on that new version please open a new issue.