darthcloud/BlueRetro #202: Control DS3, always starting as 4th player

NicoBren, 2022-03-04:
Good afternoon! I bought a new PS3 controller today,  it connects to BlueRetro normally, but always as 4th player. I already tried Factory reset. Thanks for the help in advance.

darthcloud, 2022-03-04:
Does the input work?

Is it only a LED issue?

NicoBren, 2022-03-04:
Inputs are not working.

darthcloud, 2022-03-04:
Is it an official Sony DS3?

Connect your ESP32 via USB (no need to wired to console) and get the log:
https://github.com/darthcloud/BlueRetro/wiki/Getting-BlueRetro-debug-logs-via-Serial-port-Windows-10

NicoBren, 2022-03-04:
It's not a Oficial.

Log: https://pastebin.com/ZDhr1wsv

darthcloud, 2022-03-04:
Ok I don't mind helping you but you need to elaborate in your answer a little, if it's not an official one, then tell me the brand it is and the model.

Base on the log it look like a Nintendo Switch controller.

NicoBren, 2022-03-05:
On Windows it is returning the official control ID (HID\VID_054C&PID_0268&REV_0572) however, it is one of those unbranded. I have 2 more unofficial DS3s and yes these don't even connect to BlueRetro. Only the one that in Windows appears as an official control without being. Searching for the barcode, the brand appears to be Xzhang.

darthcloud, 2025-01-24:
I made a new to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

closed 2025-04-16 by darthcloud