darthcloud/BlueRetro #224: N64 3V3 Pin Goes to 5V pin on ESP32

pmgducati, 2022-03-14:
Hi,
I was working on the N64 version and was having trouble connecting.  The BLE Pairing light was not breathing and was very dim.  After looking over your schematic and doing some probing I noticed that you have 3V3 going to pin 7 on the DB35 which goes to the 5V input.  I believe it should go to pin 6 to go to the 3V3 input.  Once I swapped it everything works as intended.

[image]

darthcloud, 2022-03-14:
Yeah my own version got a low voltage drop LDO on the base, so I feed N64 3.3V through the VIN pin. This come up before, I need to update that schematic.

darthcloud, 2022-03-15:
I updated the N64 schematic with an option for my own board, But otherwise it should be ignored and wired as you draw.

Thanks!

closed 2022-03-15 by darthcloud