darthcloud/BlueRetro #225: PSX PS2 LDO is obsolete

pmgducati, 2022-03-14:
The LDO that you specify for the PSX/PS2 is obsolete - 
https://www.digikey.com/en/products/detail/diodes-incorporated/AZ1117EH-5-0TRG1/5001328

darthcloud, 2022-03-14:
@pmgducati so which part you endup using yourself?

pmgducati, 2022-03-15:
none, I'm using a ESP32 Devkit so I put the 8V to the 5V pin as stated in the Schematic.  I was looking at the SMD BOM and just googled it out of curiosity.

darthcloud, 2022-03-15:
ha ok , I'll look for replacement then, thanks to bring this up!

pmgducati, 2022-03-15:
The List is short and in the wonderful world of shortages nothing is available ATM:

https://www.digikey.com/short/dqfv1wwp

Funny thing is that is why I went the through hole route with my PCBs, the surface mount SN74AHCT1G125DCKR were nowhere to be found when I started (as of now there are only 266 in stock at Digikey)

darthcloud, 2022-03-15:
Going down to 0.8A output bring some option:
https://www.digikey.ca/en/products/detail/analog-devices-inc/LT1117IST-5-TRPBF/959151

Same footprint, I updated the wiki.

Thanks again!

closed 2022-03-15 by darthcloud