darthcloud/BlueRetro #228: Dreamcast VMU

dragoncity17, 2022-03-17:
First of all, thank you for BlueRetro it's just great! :)

I'd like to emulate a VMU for the Dreamcast, when I look at the github I see : "Memory: TBD"

What is TBD ? :)

I hope there is an possibility, I can test without problem.

Thanks in advance ! :)

darthcloud, 2022-03-17:
TBD = to be done

So not yet implemented and I don't have a time frame on when I will do.

closed 2022-03-17 by darthcloud

dragoncity17, 2022-03-17:
Ok thanks for quick reply.

I can test, so if you find a little time just reply :)

It will be amazing if VMU can be added ! 

Because for example PS2 or GameCube not have this problem, can put memory card.

But for VMU need plug directly on controller... so with RetroBlue need emulate VMU :)

Thanks in advance darthcloud ! :)