darthcloud/BlueRetro #236: Intermittent connection in N64 multiplayer games

DeathHorse1, 2022-03-23:
When I play Smash Bros and connect a wireless controller as player 1 and a cord N64 controller as player 2, the player 1 can't move the cursor correctly, this reset the position on second later.
The same issue occurs if I connect a wireless controller in other socket and the next controller has been connected by cord.

darthcloud, 2022-03-28:
How are you setup? Are you connected externally using one port only?

DeathHorse1, 2022-03-28:
I connected the ESP32 internally, for the 4 controllers. When I was playing Smash Bros, I played with a PS4 controller for the player 1, in the port 2, I connected a N64 cord controller for the player 2.

darthcloud, 2022-03-28:
This not supported.

BTW this pure luck if it work for some people. Results may vary base on various factors.

My official stand is that you can't use BlueRetro with original controller simultaneously. You either need switch on ESP32 power or on individual data lines.

closed 2022-03-28 by darthcloud