darthcloud/BlueRetro #237: Doesn't turn on esp32 externally on ps2 Fat SPHC 300001

khtp, 2022-03-24:
Does not work externally (outside the console without opening it) ps2 fat model SPHC 30001 and SPHC 39000 models, the esp32 does not turn on, it has already been verified that the connector of both consoles with original controls work and if they work they vibrate without problem in ps2 slim SPHC 7000X if they work
[image]

darthcloud, 2022-03-30:
Just to be sure, is your ESP32 GND wired to the PS cable GND. and VIN is wired to to PS 8V?

I ask because somehow people using this specific module look to follow some guide that omit the power connection (instead of my wiki)

Alternatively, try to remove 8V to VIN connection and instead connect ESP 3V3 to PS 3.3V.

pmgducati, 2022-03-31:
I can confirm @darthcloud suggestion on switching over to 3.3V.  I have a SPHC 30001 and I was getting power but the ESP would not map a controls, once I switched over everything has worked perfectly.

khtp, 2022-03-31:
@pmgducati do you use esp32 internally in your ps2 fat or externally? @darthcloud I was testing the photo that I published and I realized that the VIN pin with that yellow cable had no continuity, in conclusion it did not work in slim or fat, that was the error that I had, I had to eliminate it and now I am using the micro usb to power the esp32 works on the slim, now I followed a tutorial on a youtube channel so I didn't connect pin 3,4,5 of the connector because the youtuber I was following told me not to connect it, ask : Is it necessary to power the esp32 using pins 4(GND) and 5(3.3.V) through the ps2 connectors for it to work on the ps2 fat, is that the only way?

darthcloud, 2022-03-31:
Try it and see if it work.

pmgducati, 2022-03-31:
@khtp my setup is external, I designed some PCBs based off @darthcloud's schematics.  I was using the 8V in on my PCB and in my case I was able to get power but no controls (the debug thought it was a PC-FX).  Once I disconnected the 8V pads and connected pin5 of the controller to 3.3V, everything worked without an issue.  

Here are some pictures of my setup:
This is the new version that includes both 8V and 3.3V pads:
[image]
[image]

Here is the PCBs I have now with the 3.3V rework:
[image]
[image]

As for the youtuber pin 3/4/5 recommendation.  If you are powering via USB you will still need Pin 4 connected (GND) for it to work reliably if at all.  In order for it to work at all without the USB you would need either pin 3 or pin 5, but not both.

I updated my PCB to have both per @darthcloud's recommendation as he said that some PS2's work with one but not the other and I wanted to make sure that if anyone happened to grab my PCBs they would have the option if their PS2 was finicky either way

Hope this Helps!

khtp, 2022-04-06:
It already worked with ps2 fat my problem was that the VIN connector of the esp32 with pin 3 did not have continuity, it was badly soldered, I solved it by eliminating that cable and now I am feeding it with the micro usb to the usb port of the console, now I have a new problem, if I play more than 1 hour out of nowhere the game tells me to connect player 1 and restart esp32 and it works again and if I leave the ps2 on and without moving anything for 5 more minutes it tells me connect the player 1 again, that is, it tells me controller disconnected, I don't know if this detail can be solved? And by the way thank you very much for your advice if they helped me a lot and this project turned out great for you, thank you very much for contributing this wonder because now you control bluethooht that I had saved now I am a use for it

khtp, 2022-04-12:
@pmgducati  Good evening, by any chance, did you try the 2nd generation xbox controller version 5.1 and have you played for more than 1 hour? Haven't you noticed if out of nowhere player 1 disconnects out of nowhere? That is happening to me on the ps2 fat sphc 30001, if I remove the corroente and turn on the espe2 again it works for me but the same thing happens again, and thank you very much for the advice if it helped me, my solution was to remove the VIN and the pin 3 and I'm feeding it with the micro usb and if it works for me, on another console like gamecube that doesn't happen to me

pmgducati, 2022-04-12:
Hi @khtp ,
I have had no issues, I am use 2x 8bitdo SN30 Pro+ controllers and have had nothing but success with all the systems that I have made PCBs for the Blue retro project.  

https://www.8bitdo.com/sn30-pro-plus/

khtp, 2022-04-21:
@pmgducati  @darthcloud 
Thank you very much to both of you for helping me and especially to you @darthcloud for carrying out this project and sharing it and an apology for being insistent.
 
I already solved my problem: if I use pin 4, that is GND and connect it to the GND pin of the esp32 and eliminate pin 3 and 5 that would be 8v and 3.3v.

Although you told me not to remove from the flash download tools the one that says: DoNotChgToBin < in my case I removed it and updated to FW 1.5.1 and my problem was solved that said: that the controller is disconnected during the gameplay and I was playing more than 2 hours and I had no problem with my xbox one x fw 5.1 controller, in my case I am feeding the esp32 through the micro usb to the usb port of the ps2, just remove pin 3 and 5 from the connector and that's it.

another note: I have the OTA version by default

I hope what I said serves as a guide or you see the reason why removing that option from the flash download tools works and will solve my problem

If I have any other problem or the same problem keeps appearing, I'll let you know.

darthcloud, 2022-04-21:
If you used OTA once, that version will always take precedence over anything you flash via USB.

If you want to use the version flashed via USB version you must factory reset the adapter by holding boot 10 seconds.

khtp, 2022-04-21:
no never use it, I normally use the flash download tools to update to the latest version of the FW which is 1.5.1 in your latest release

closed 2022-05-07 by darthcloud