darthcloud/BlueRetro #251: esp32 error on ps2 after 5 minutes

DonCristobal2021, 2022-04-13:
Good afternoon

I have an error with an original ps3 controller on the ps2 with blueretro, it seems everything works fine at first, but after a while, about 5 to 10 minutes, it starts to go crazy, it doesn't recognize buttons, or it lags a lot, I have to disconnect the esp32 and reconnect it so that it returns to normal, I connect the esp32 to the usb port of the ps2, will you have a solution for this please.

darthcloud, 2022-04-14:
try what is suggested in issue #250

khtp, 2022-04-14:
are you using pin 4 of the connector which is GND? if you don't have it, then connect pin 4 connector to the GND of the esp32 and it should last longer, it happens to me with the xbox one 2 generation controller, it lasts 1 hour well and then the console asks me to reconnect the player 1 I have FW in 1.5, which FW are you using?

DonCristobal2021, 2022-04-15:
If I have GND soldered to the corresponding pin, right now that dathlogan sent me, there is someone who said to solder pin5 to 3v3 of esp32, right now I did that, if so, it no longer has problems during the game, but I will check well how it works

khtp, 2022-04-21:
I already solved a problem, as a recommendation disconnect pin 5 3.3v and feed it by micro usb and update the FW to 1.5.1 in my esp32 it already works without problem and it does not turn off I am using a ps2 fat sphc 39000, reading what this does esp32 board by default needs 5v minimum 1 amp and the pin 3 connector only gives 8v and 3.3v although the esp32 board supports up to 15v because it has a converter, this converter converts it to 3.3v I understood that and I can be wrong until they correct me , I was afraid to connect the 8v to the VIN pin of the esp32 and better connect it via micro usb and to date it has never been turned off playing that was my problem also in the flash programmer donwload tools remove the check mark that says: DoNotChgToBin << <I did have it activated and it gave me problems like the controls disconnecting, I disabled it and it hasn't given me any problems, I've only noticed that the esp32 gets a little hot but because where I am it's very hot or I've been playing more than 2 hours

closed 2023-04-01 by darthcloud