darthcloud/BlueRetro #254: 3do esp32 version works only with usb power supply

scp-electrocution, 2022-04-17:
I made the 3do version of the bt receiver on esp32. It works well by powering the chip from usb while powering the esp from the console does not work, the chip is powered correctly and the led lights up.
D18 and D19 are bridged
I attach log

[log3do.txt](https://github.com/darthcloud/BlueRetro/files/8501903/log3do.txt)

darthcloud, 2022-04-17:
@RobertDaleSmith had the same issue until I build one for him. Can you confirm what was the issue?

I think the issue was the db9 connector not passing power properly, i'm using genesis cable extension btw.

scp-electrocution, 2022-04-17:
I connected the 2-5 + 5v and 1-8 cables for gnd, i tried to connect a 12v-5v step down but the problem was not solved, it only works with usb power

scp-electrocution, 2022-04-18:
I solved the problem, the (Chinese) cable was too long and has tiny conductors