darthcloud/BlueRetro #259: Not able to pair controller or connect via web interface on N64

jon-nielsen, 2022-04-18:
Thank you for this project first of all.

I'm creating this issue even though I'm not sure if it's a problem with the BlueRetro code or the N64 itself.

Anyhow. My build is a simple cable build using an N64 extension cord. The cord worked for connecting a standard controller with. 

3v3, gnd, data are connected to 3v3, gnd, D19 on the ESP32.

D39 is jumped to 3v3. I've tried doing the same for D5, D26, D27.

I've also tried connecting 3v3, gnd to vin, gnd. The results are pretty much the same:
The light turns on, but pairing is not possible. My PC does find device eventually when powered on via the N64, but it is never able to make a connection for the web interface.

Pairing the controller is however possible when connecting the adapter to USB instead of the console. But it doesn't help when switching over to the N64.

My thoughts are that the 3v3 rail on the N64 is either too weak or too noisy, if that makes sense.

So I'm creating this issue in case anyone has experienced the same and can think of a solution.

Thank you.

darthcloud, 2022-04-19:
Post a picture where we can see connection well.

darthcloud, 2022-05-24:
You could try adding a 1uF and 100uF cap between the ESP32 3V3 and GND

jon-nielsen, 2022-06-08:
Hey

I've tried now with 2 ESPs on a couple different N64s and various caps on 3.3v: 470uF electrolytic, 1uF electrolytic + 100nF ceramic (don't have any of the types you mentioned exactly) as well as an experiment where a 5v step up regulator was used to provide power to VIN. Result is the same. Anyway. Here is a picture of current wiring, no caps.

Cheers!

[image]

AgustinCingolani, 2022-06-11:
Hi @darthcloud, I'm having the same or a similar issue.
Connecting to N64 (Pins 3v3, GND and IO19 just for P1) I'm not able to pair the controllers. But if we plug the usb cable (disconnected from the N64 of course) I can.

What do you suggest?

Thanks in advance

delislej, 2022-06-18:
> 

did you connect the rest of the p2-p4 pins to 3.3v? it says in the wiring diagram that you need to wire the unused player IO pins to 3.3v or weird stuff will happen.

[If not wiring 2nd port plug, make sure to connect IO5 (P2_D) to 3.3V
If not wiring 3rd port plug, make sure to connect IO26 (P3_D) to 3.3V
If not wiring 4th port plug, make sure to connect IO27 (P4_D) to 3.3V
](https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-Cables-Build-Instructions#nintendo-64-adapter-cable)

maybe that might help, i know i ran into issues until i did that.

AgustinCingolani, 2022-06-21:
@delislej thanks for your suggestion, but here the issue is that I can't pair the controller to the ESP32. 
This seems to be a power supply issue, maybe with only 3.3v this board (Is a clone) does not supply correctly the bluetooth chip.

Are you sure with just wiring IO5, IO26 and IO27 maybe solve this issue?

darthcloud, 2022-06-21:
If those pin are float and are unstable they might generate an infinite amount of interrupt that will crash in loop the adapter.
Which will prevent BT connection.

Wire all 3 pin to ESP 3.3V pin.

AgustinCingolani, 2022-06-21:
Okok, thanks a lot to both for the suggestion, I'll try and then let you know :)

jon-nielsen, 2022-06-23:
Quick follow-up. I've updated the firmware to latest version, updated wiring like so, and confirmed that I can still pair a controller when powered by usb. Same trouble when running from the N64, though.

[image]

darthcloud, 2022-06-23:
Try with an official ESP-DevKit-C  (Get it on [Digikey](https://www.digikey.com/en/products/detail/espressif-systems/ESP32-DEVKITC-32E/12091810?s=N4IgTCBcDaIKYGcAOBmMBaAJnAbgawEsAXAY3TThAF0BfIA) or [Mouser](https://www.mouser.com/ProductDetail/Espressif-Systems/ESP32-DevKitC-32E?qs=sGAEpiMZZMv0NwlthflBi3XYxq%252BTapwyHnR2%2FrAf4bY%3D))

delislej, 2022-06-24:
I actually ran into this exact issue with the first ESP32 I got off amazon, it was clearly a clone. I got around it by using a USB battery bank to supply the 5v, but eventually caved and ordered my ESP-DevKit-Cs off of digikey.

jon-nielsen, 2022-06-24:
I will try that once my trusted supplier has them in stock again (September or November). The clones aren't cheaper, but they have worked before for BlueRetro builds. This is the second ESP32 that doesn't work properly, though. Any idea what the typical problems with these clones are?

AgustinCingolani, 2022-07-02:
Didn't work. I'm going to try with the Espressif chip and let you know.

AgustinCingolani, 2022-07-05:
I've tried with an original board but were the same result. What do you suggest?

darthcloud, 2022-07-05:
No idea at this point, sorry :(

novamaxrc, 2022-08-01:
Hello
I think it is related to issue #146, it didn't pair with a single controller with just n64 but it paired when connected to usb, then i connected the second port to vin, p5 and gnd, so now the first controller pairs fine and works brilliantly, the second one hardly pairs and when it pairs it have massive lag. i then connected p26 p27 to vin but same results. i made many tries connecting all powers and p26 p27 to 3v3 or all to VIN and nothing paired, now it works just the first controller with both controller power to 3v3 and p26 p27 to VIN. The second controller has the same heavy lag even if esp32 is connected to usb

jon-nielsen, 2022-08-01:
I've received a bunch of ESP32 devkit boards off AliExpress that just work. In the meantime I've had a chance to test my old batch of boards on other consoles with various degrees of stability (NES, GC, PSX). New work just fine, instantly pairing and not losing connection, all with default wiring. I think we can close this issue and just conclude that some boards are of poor quality.

darthcloud, 2022-08-01:
@jon-nielsen would you mind shipping a few of your bad batch of ESP32 to me?

Ping me on discord DarthCloud#9363 or twitter DM

novamax89, 2022-08-01:
> I've received a bunch of ESP32 devkit boards off AliExpress that just work. In the meantime I've had a chance to test my old batch of boards on other consoles with various degrees of stability (NES, [...]

Hello. Thanks for the information. In your opinion how many have I to order to get one working? Are 5 enough? I will buy them from Amazon so I can send back the non working ones. Thanks

soniccd123, 2022-08-04:
Hello, i have the same issue, my boards actually keep reseting continuously. I have tested it thoroughly and it seems to me that some ESP32 boards consume more power than the N64 controller ports are capable of delivering. In a test, on the USB the board's 3.3V rail measures 3.36V on my multimeter. Powering the board through the controller port I get 3.24V. Note that the port supply measures 3.33V with an original controller or nothing connected. This shows clearly a voltage drop that I can only conclude happens because the N64 ports can't supply the needed current to some of the clone ESP32 boards. (Unfortunately the ammeter in my multimeter is not working correctly for a more direct test)

I have not tried to remove the CP2102 UART and/or the 3.3V regulator (as the N64's 3.3V rails are obviously already regulated) to see what happens, but maybe this lowers the power consumption enough to get it going.

While this sucks for now, my plans originally where always to have the Blueretro adapter installed internally on the console. Inside the N64, instead of soldering the ESP32 to the controller port 3.3V rail, I connected it to the console main 3.3V rail, directly from the power supply, right after the power switch (if you connect it before the ESP32 will always be on). This works like charm and also confirms to me that the 3.3V rail from the ports cannot supply enough power (the main 3.3V rail can supply 2.7A if my sources are correct, which is more than enough to power the console + an ESP32)

jon-nielsen, 2022-08-04:
> Hello. Thanks for the information. In your opinion how many have I to order to get one working? Are 5 enough? I will buy them from Amazon so I can send back the non working ones. Thanks

I bought 6 because I intended to build 4 units. The first 4 worked. I can't speak for the quality of all AliExpress sellers. But what I got was better than the 2 out of 3 failure rate from a local shop.

jon-nielsen, 2022-08-04:
> Hello, i have the same issue, my boards actually keep reseting continuously. I have tested it thoroughly and it seems to me that some ESP32 boards consume more power than the N64 controller ports [...]

For what it's worth, I was focused on similar observations. With the new (cheap) boards I got off AliExpress it's not an issue. They just work. Maybe you could try another board.

delislej, 2022-08-04:
> Hello, i have the same issue, my boards actually keep reseting continuously. I have tested it thoroughly and it seems to me that some ESP32 boards consume more power than the N64 controller ports [...]

Some N64 power supplies are failing at this point. My controller ports put out a clear 3.3v, but I bought a non-OEM PSU since my older one was failing.

novamax89, 2022-08-05:
Hello. I just tried with 2 different esp 32 boards and the problem is solved. I confirm at this point that the problem is related to a bad manufacturing of the board.

damontgomery, 2022-09-20:
First, thanks for the project, very cool!

I think I'm having a similar issue.

I was able to get this to work with
- ESP32 Devkit C (Espressif brand from Mouser)
- USB power while connected to n64
- 1 controller
- n64 specific bin downloaded today, flashed on Win10
- players 2-4 connected to 3.3V
- ESP32 on a breadboard
- Extension cable with about 8" of length soldered to jumper wires plugged into breadboard
- Optional red LED connected
- PS4 controller
- Mario64 original cart in system

When I had the same setup but with 3.3V from the controller plugged into the 3.3V line on ESP32 and no USB power, the LED would turn on, then off and blink about once every 10 seconds. I think at the same time the GPIO 19 data line was high for the whole time and then would drop low at the same time the blinking happened. The controller would not pair. In this setup, I think I had about 2.7V across 3.3V to GND, which seems low. Maybe my power supply is dying?

I'm very new to using a multimeter, etc and don't have an oscilloscope or anything to provide anything much more detailed than this.

I was also able to get something similar working with my PS1 without USB power and the controllers connected no issues.

For now, USB connection to power the device is fine. Up for trying some things if you have suggestions.

I just started with this which is why I've got it all on the breadboard. I may eventually solder this together. I have 2 of the same ESP32, and may try the other if that could make a difference.

Thanks again.

darthcloud, 2022-09-20:
Try to get it off the breadboard, this often create problems.

damontgomery, 2022-09-21:
> Try to get it off the breadboard, this often create problems.

Thanks for the information. I may revisit this later and see if I can get that to resolve the issue.

damontgomery, 2022-09-22:
TLDR: Powering over USB may cause video noise and may also cause the console to reset. I'm decently confident of the noise issue, and not very confident on the reset issue.

---

Don't want to spam the thread, but wanted to provide an update on my "work around" of powering over USB.

I had a number of issues and not sure which of these if any is related to the device. I realize a lot of this is not very specific, well researched, or reproducible, but maybe someone has a similar issue and this will help them.

When powering over USB and plugged into controller 1 (with the setup above, breadboard, etc), I got video interference that looked like several static / wave noise across a horizontal section of the screen. Without the device plugged in, or plugged in to 3.3 via connector, this wasn't there.

The part I'm less certain of is that my n64 became unusable since it would restart every second or every few minutes while I was trying different setups.

I was able to eventually resolve it with:
- Unplugging the power adapter from the wall and the console
- Unplugging the jumper pack and replugging
- Unplugging the cartridge, cleaning, reseating
- Using an original n64 controller

No idea if any of those specifically helped, but I was then able to play through the first hour of The Legend of Zelda Ocarina of Time.

I had tested LoZ:OOT right before with the USB power and it seemed to work fine, I played for a few minutes and could make a save. Then, I replaced the battery (tested it had over 3V), cleaned, reassembled, then started having issues.

Mario 64 didn't seem to have any of those issues, but I never tried it for more than a handful of seconds because at the time LoZ wouldn't work for more than a second or two.

darthcloud, 2022-09-22:
You need to disconnect the 3V3 from n64 if you power BlueRetro externally.

Only plug data and GND.

damontgomery, 2022-09-22:
> You need to disconnect the 3V3 from n64 if you power BlueRetro externally.
> 
> Only plug data and GND.

Thanks! When these issues were happening, I only had data and ground connected.

delislej, 2022-09-22:
I remember having similar issues when I ended up with some clone boards off
Amazon. It wasn't powering off just the controller port, but using USB 5v
it was causing problems with the n64 that almost broke the system. I was
getting random restarts as well. I got my current boards off digikey and
tossed the crap clone boards.

darthcloud, 2022-12-14:
I think the consensus here is that cheap board perform poorly especially on n64 likely due to its tricky 3.3v supply. Closing.

closed 2022-12-14 by darthcloud

soniccd123, 2023-03-19:
I know this issue is closed, but I have found a cause for this problem that maybe the same for others here. In my case, i've already explained in detail my setup in and older post, I've found a problem with my setup: I was connecting the MCU to a cable that i had made myself and, in further inspection, I found that, while the cable was making contact to the N64 connector, it had bad contact in 3.3V ou GND pins (maybe both), so the cable was not correctly made, contact resistance was high and the MCU could not get enough power to work. Connecting the MCU directly to the internal contacts of the N64 ports made it work correctly. Making a new cable using adequate technique made it work too. Tested with 3 ESP32 versions (all chinese knock offs) and all 3 worked. 

So TLDR, check your contacts, even if you get the MCU to partially power on, a bad connection may be the culprit for this problem, if you're not sure, try to connect the MCU to the internal pins of the connector, this will give a good indication of your connector status and connection quality.