darthcloud/BlueRetro #262: START Button on XB1 controller not working

bixxewoscht, 2022-04-28:
Note: I could not test this myself yet.

Several people told me that the Start button (3 lines) on the XBOX One controller does not work with the default mapping.

It is a quite common controller, so it would be able to use it without entering the web config first.

I'm sure many people would appreciate an update in a future release, if you have the time.

darthcloud, 2022-04-28:
They just need to updated their controller to latest FW with the xbox
accessories app.

I only support latest xbox FW as noted on wiki:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#xbox-one-s-series-xs--adaptive-controller

darthcloud, 2022-05-07:
Reopen if FW update do not fix the issue.

closed 2022-05-07 by darthcloud