darthcloud/BlueRetro #264: Nintendo 64 Controller Pak / F-Zero

Taurus1099, 2022-05-01:
I’ve enabled the controller pak manager through the web interface and formatted it but a number of games prompt that there is a problem with the pak on launch. One example is Tony Hawk Pro Skater 1. As a separate issue, F-Zero does not detect the blueretro device as a controller. 

Console is modded with N64 digital and using latest version of the blueretro firmware. Unrelated, but with two NSO 64 controllers the number returned from the controller test tool for the right axis seems to be a bit higher than the benchmark. What setting in the advanced config should I adjust?

darthcloud, 2022-05-02:
Try to format via a laptop if you used a phone, or just let tony hawk format it when prompted.

If detection fail you can simply press reset on console, also games give up too fast.

Play with the LX, RX maximum value in web config to fine tune joystick.

IMHO this is a waste of time, that test tool make people paranoid. Most games are programmed with a good play room.

Taurus1100, 2022-05-05:
Thanks for replying! I've tried the suggestions but F-Zero does not work at all with a blueretro device and Tony Hawk will not recognize the controller pak. Testing a random sample of other controller pak games shows a major issue with the current integration. I've also experienced odd rumble behavior. On Goldeney 007, the rumble will not work correctly when firing. It gives a brief rumble and then stops when firing, when it should continue to rumble when firing.  This is a great product, and hopefully you'll be able to circle back around to the N64 to finish polishing the feature set. Thank you!

darthcloud, 2022-05-05:
You might have some build related issues because I personally got none of those issues.

Sound like your data line is glitching somehow.

Taurus1100, 2022-05-05:
This was confirmed from others using the adapter purchased from Tindie so it is likely the build. I’ll follow up with the seller.

darthcloud, 2022-05-05:
If you can ask others to report feedback in this GitHub issues.

Everyone with issue should list the following information if possible:
* N64 motherboard revision
* List of internal mods installed
* Other external peripheral used

It could be problem with the dongle itself but it could also be some interoperability issue with some motherboard or some other mods.

Taurus1100, 2022-05-05:
@bixxewoscht

bixxewoscht, 2022-05-05:
I had mempak issues myself before, but they were resolved when I used a PC instead of a phone to change the config. These issues occured both on my adaptor as well as on an ESP32 Devkit which I used to compare functionality.
Regarding the other issues, both with a ESP32 Devkit as well as my own Adaptors I had sporadic rumble issues, close to the ones that you described. 
Nothing gamebreaking though, the core functionality always worked. To be honest I always accepted it as minor issues in a great open source project, maybe depending on different controllers, console revisions and mods used. In the issues section on GitHub there are also several topics related to these small issues, so I did not look more into it and accepted it as small software bugs that don't really affect me.

Now of course I don't want to blame you @darthcloud for anything in your amazing project and I want to offer the most for everyone who bought my Adaptor, so I will look further into it. 

Because I just moved last week, it will take more time than usual, I barely even have a TV set up at the moment, and my N64 consoles as well as the Adaptors are still in boxes.

I will try to get to it this weekend.

darthcloud, 2022-05-05:
@bixxewoscht Anything is possible off course but the N64 is the original system BlueRetro was developed for and that I actually use myself a lot. I gone though GoldenEye a few time ever the last 3 years and never noticed any rumble issues. The way it is implemented makes the interface very robust as it is using the ESP RMT peripheral, (so no software bit bang). What you guys describe should not happen at all.

I'm using my own BlueRetro Devkit when playing  but I did test with a few clone ESP-DevKit-C in December and this spring and didn't got any controller pak error myself. Controller pak was tested by going through Turok 1 and Tony Hawk 1.

The issue sound like the data line is glitching.

My own board got ESD protection on each line using a TPD6F002DSVR chip. This got an equivalent resistance of 100ohm. No idea if this help or not.

Maybe chip me one of your dongle for me to test.

I play with early N64 rev mostly but got a few more now that I could try on too. I got no mod in my console.

darthcloud, 2022-05-17:
@Taurus1100 / @Taurus1099 I got your adapter today and made some quick test and it look to be working fine for me.

See video:
https://youtu.be/IBnAhmHv9O8

I tested with your F-Zero X and TH1 cartridge with a NUS-CPU-04 and NUS-CPU-08-01 motherboards. Both work fine.

I got others N64 mobo in storage will dig them this weekend and report back.

Meanwhile can you confirm which mobo your N64 got? Did you install any mods? Are you using an original AC adapter?

taurus1101, 2022-05-18:
Thanks for taking a look. My console revisions is NUS-CPU-08-01 (modded with N64 Digital) and I'm using the original power supply. I also tested with an additional power supply and no change. When you are able to do further testing, are you able to get into a game on F-Zero? I've seen the game respond to controller inputs for 10-15 seconds but eventually it stops completely. The N64 Digital menu is still accessible however. Also, any issues with the rumble in Goldeneye?

darthcloud, 2022-05-18:
I think its likely the N64 digital that interferes with the pad data lines. I can test more this weekend but its likely all fine.

I don't know much about N64 digital but I think you could remove the one connection for pad and it would likely work fine. But you would lose the OSD access.

darthcloud, 2022-05-18:
And what's up with using new account every comments?

taurus1101, 2022-05-18:
Must be related to the N64 Digital. I noticed a similar issue on another open source project: 

https://github.com/Ryzee119/usb64/issues/42

bixxewoscht, 2022-05-19:
I tested with my N64 with N64 Digital and had the same issue with F-Zero. Could not test with my stock N64 yet. But in my other tests I remember there were no issues with the game on a stock N64.

Thank you both for investigating further.

Holen Sie sich Outlook für Android<https://aka.ms/AAb9ysg>

Ryzee119/usb64#42<https://github.com/Ryzee119/usb64/issues/42>

darthcloud, 2022-05-20:
[n64_test_push_pull.zip](https://github.com/darthcloud/BlueRetro/files/8740052/n64_test_push_pull.zip)

This is untested but it change the data line from open-drain, to push-pull. Maybe that will help if the issue is on the TX side.

taurus1101, 2022-05-20:
Everything is now working perfectly! The added ability to get vibration feedback on switching between rumble/memory pak is very helpful. Is there any functionality lost by changing the data line?

astro187, 2022-05-20:
Hi @darthcloud, I'm the guy that slid into your DM's.  Similar to the above user, I can confirm that the new firmware fixes all of the issues I was having. I played with it for quite a while today and exercised all of the main functionality and everything worked great.  Thaks for finding a workaround so quickly... I was excited, then super bummed, and now excited again... this is a really neat implementation and will be my goto N64 setup going forward!
________________________________________________________________________________________________________________
Just in case it is useful info for future reference, when I _was_ having problems it was the following setup:
NUS-CPU-08-01 with N64Digital mod
stock PSU
Switch Pro controller
v1.5.1 of your firmware
@bixxewoscht receiver

What was happening (this was only for P1 on my modded system; P2 worked fine and my stock NUS-CPU-08-01 worked fine)
Controller pak emualation wasn't working at all (games either wouldn't recognize it or would report it corrupted)
Every few seconds an uncommanded controller input [Dpad  down + Dpad right + main joystick (1,-1)] would register

mario-64, 2022-05-23:
I can confirm this firmware seems to resolve the issues I was having on my n64digital console as well, both with F-Zero X and overall memory pak issues.

darthcloud, 2022-08-13:
I got an N64 digital this week and was finally able to investigate first hand.

I wasn't really a fan of the workaround I made to fix it, the N64 SI is means to have device with open drain IO. So switching to push-pull is not really ideal IMHO.

This is really an issue caused by the N64 Digital mode and I think the fix should be an HW on it.

To fix it I added a 74AHCT125N buffer between the the PIF pin 16 and the N64D ribbon COM pad

This did the trick F-Zero X menu doesn't glitch with BlueRetro and the N64D OSD still work as before.
[image]

The best would be to design a ribbon with a pad for the tiny 1 gate variant 74AHCT1G125W5-7

[image]
[image]

mario-64, 2022-08-13:
Will you please leave the software fix in place for those of us without skills to mod our systems? I would have to pay for this work to be done :-(

darthcloud, 2022-08-13:
Problem is this fix doesn't work for everybody, and it actually break it for some people without a N64D.

Old FW stay available so you can keep what you currently have, and if any major update occur I can always post a special version in this thread

mario-64, 2022-08-13:
Great thank you

darthcloud, 2022-10-08:
Quick update PixelFX designed a small flex to fix this issue. I tested it and it work great.
[image]

I'll update this issue once info on how to get it is available.

timotchee, 2022-10-08:
@darthcloud Is this fix the same for both HW1 and HW2? ie for both external and internal (with detection) installs?

darthcloud, 2022-10-08:
@timotchee Yes

timotchee, 2022-10-08:
@darthcloud If I'm reading correctly, you prefer Dan's flex cable fix to your 74AHCT125N buffer method for electrical reasons?

darthcloud, 2022-10-08:
They are both using 125 buffer, it's same thing except flex look nicer

darthcloud, 2022-10-10:
Flex available here: https://www.pixelfx.co/n64digital-controller-flex-w-buffer.html

BassXX, 2022-10-12:
Hello there, is there also a fix planned for UltraHDMI users? I updated to the latest firmware and now the system does not detect the controller at all anymore

Sowden, 2022-10-17:
Yikes!  I have just ordered one of the dev kit chip to do an internal install on my UltraHDMI modded N64.  Has anyone reported a similar issue bundled with that mod?  If so, a new flex cable will have to be developed, and as nice as Marshal is, he's only one guy and has a busy life.  The chances of him developing one is slim, unless someone else steps in and develops one of their own.  But again, has anyone with a UltraHDMI mod had this problem?  Should this deveate into a different thread?

pixelherogaming, 2022-12-02:
> Flex available here: https://www.pixelfx.co/n64digital-controller-flex-w-buffer.html

This flex doesn't seem available anymore?

I am having an issue with my N64 digital modded system along with the blue retro external mod. My NSO and 64 controller will sync. And then randomly will disconnect mid game. Any work around?

timotchee, 2022-12-02:
Dan of pixelfx/blackdogtech said there will be more flex cables in the future, but without a date.  I would guess in the next 5 months.  Or you can build your own circuit w/buffer with the info above.  If you go the FFC route, you will need to solder surface mount components onto the FFC.  Below I've listed the component to DIY, and the SMD components Dan originally specified for his flex cables, but he removed the FFC web pages so it's possible they change in the future.  There's no difference between the two, the FFC simply looks nicer if you're into staring at the innards of your console.  The DIY option will be easier to solder.

Buffer for DIY circuit - TI SN74AHCT125N: https://mou.sr/3ivItle or https://www.digikey.com/short/9z5dz7pp
FFC N64 Buffer - Nexperia 74AHC1G17GVH: https://mou.sr/3gPJYKN or https://www.digikey.com/short/p7jhvrff
FFC N64/PS1 Ceramic Capacitor(s) - 0.1uF 10V 0402: https://mou.sr/3uhdFaF or https://www.digikey.com/short/qt4nq3bq
FFC PS1 Buffer - TI SN74HCS125QPWRQ1: https://mou.sr/3H6pdoF or https://www.digikey.com/short/0mq77pbf

Please double check above and the FFC listing whenever it returns to ensure I made no mistakes.

pixelherogaming, 2022-12-02:
> Dan of pixelfx/blackdogtech said there will be more flex cables in the future, but without a date. I would guess in the next 5 months. Or you can build your own circuit w/buffer with the info above. [...]

So I should just need to order the buffer for diy (since i dont care what it looks like) and them just solder like example above? Or do I also need capacitor(s)

timotchee, 2022-12-03:
Yeah, DIY is easiest and fastest.  I don't see any caps in the circuit, but @darthcloud is the expert, if he sees this.

mario-64, 2022-12-03:
My recommendation is to avoid firmware 1.8 and above or, alternatively, stop using BlueRetro for N64. We shouldn’t have to further mod our consoles.

L10N37, 2022-12-12:
I had a customer send me in the flex kit and his N64 that I put the HDMI kit in. This flex seems to kill the ability for a normal N64 controller to enter the OSD?

darthcloud, 2022-12-12:
@L10N37 It shouldn't do that, probably problem with flex install.

L10N37, 2022-12-13:
ah it was like 3/ 4am and pulled 2 all nighters in a row, i just missed the 5v connection

darthcloud, 2022-12-15:
For those that will not fix their N64 Digital install and that liked the imperfect workaround here a build of v1.8 that reintegrate it.

[v1.8-push_pull_port1_hw1_n64.zip](https://github.com/darthcloud/BlueRetro/files/10239311/v1.8-push_pull_port1_hw1_n64.zip)

pixelherogaming, 2022-12-15:
> For those that will not fix their N64 Digital install and that liked the imperfect workaround here a build of v1.8 that reintegrate it.
> 
> [v1.8-push_pull_port1_hw1_n64.zip](https://github.com/darthcloud/BlueRetro/files/10239311/v1.8-push_pull_port1_hw1_n64.zip)

Any latency or disadvantage to this work around as opposed to adding a buffer?

mario-64, 2022-12-15:
> For those that will not fix their N64 Digital install and that liked the imperfect workaround here a build of v1.8 that reintegrate it.
> 
> [v1.8-push_pull_port1_hw1_n64.zip](https://github.com/darthcloud/BlueRetro/files/10239311/v1.8-push_pull_port1_hw1_n64.zip)

Thank you!

mario-64, 2022-12-15:
The fix is integrated in firmwares prior to 1.8 but it was removed in the main branch.

darthcloud, 2022-12-16:
Its a hardware bug and the workaround only work for half people with n64 digital, and break half people without one.

The real fix is the pixelfx patch cable.

mario-64, 2022-12-16:
> Its a hardware bug and the workaround only work for half people with n64 digital, and break half people without one.
> 
> The real fix is the pixelfx patch cable.

Blueretro, N64 or n64digital hardware bug?

darthcloud, 2022-12-16:
n64 digital hardware bug, this is why they provide a flex patch

BassXX, 2022-12-16:
As an ultra hdmi user, I suffer the same problem. 

And I know for sure that there won’t be a fixed flex cable, at least not from the mod creator.

so basically I have no choice but to either switch to N64Digital or use those workarounds. I hope those will be provided in the future versions as well. Thank you

bixxewoscht, 2022-12-16:
You could also look at the Flex/DIY fix documentation and adapt it for Ultra HDMI. There shouldn't be too much of a difference. But I don't know it exactly since I've never looked at Ultra HDMI.

darthcloud, 2022-12-16:
You just need to do as I explained in my previous post in this thread
https://github.com/darthcloud/BlueRetro/issues/264#issuecomment-1214224340

Sowden, 2022-12-20:
Would the #264 post fix for those with UltraHDMI's as well?

Sowden, 2022-12-20:
@darthcloud Ok, let me see if I get this one strait.  The only wire you are intermediating is the N64D ribbon cable to the PIF 16.  You aren't messing with any of the other connections?  I'm asking because I'm finding the similarities between the N64D and the UltraHDMI.  The UHDMI has a ribbon cable that I believe you attach to the same spots as the CON connection on the N64D.  

[image]
(not my photo, taken to show example)
So would that mean that you would connect the UHDMI ribbon -> Pin 2 <74AHCT125N> Pin 3 -> (whatever pin depending on your motherboard)?  Cause that might be the solution for the select few that own UltraHDMI sets.

darthcloud, 2022-12-20:
@Sowden Yeah just piggy-back a 74AHCT125N on top of the PIF chip like I did (connect 3V3 & GND to pif pin directly) then the pin label N64D_COM (pin3 ) is the one you plug your ultra HDMI cable to. and pin 2 goes to the pif controller 1 pin. Don't forget to GND all the needed pins as done in my schematic.

Sowden, 2022-12-20:
Oh, GND Pin 1 -> PIF 14 and 3.3 v Pin 14 -> PIF 28?  So the 74AHCT125N has four connections in total, is that correct?

nortakales, 2023-05-31:
Just wanted to pop in here and see if my issue might be related. I've got an N64 with the [Gamebox Systems 64HD](https://castlemaniagames.com/products/gamebox-systems-64hd-digital-video-output-kit). When I try to use my [BlueRetro adapter (from Retrotime)](https://8bitmods.com/n64-blueretro-bt-controller-receiver-with-memory-pak-original-grey/) in controller slot 1, I am getting graphical glitches only on my HDMI out (regular composite output is unaffected). Other controller slots do not have the issue. It's almost as if (and this is just a guess) the OSD menu is being interacted with even though I haven't brought it up. I haven't tried your firmware posted here yet though, I'll give it a try tomorrow.

bixxewoscht, 2023-05-31:
From what I heard from other users (I don't have a 64HD) the Gamebox 64HD definitely has the same issue as N64Digital and other mods that tap into the controller lines. Unfortunately the known fix was not implemented from the get go on the N64HD.

nortakales, 2023-06-01:
I updated my device with [v1.8-push_pull_port1_hw1_n64.zip](https://github.com/darthcloud/BlueRetro/files/10239311/v1.8-push_pull_port1_hw1_n64.zip) and it did not resolve the graphical glitches :(

nortakales, 2023-06-01:
To be clear, with the 64HD I am NOT getting issues with F-Zero or Tony Hawk as reported here for N64 Digital and Ultra HDMI mods.

I am just getting graphical glitches when my BlueRetro adapter is in controller port one. The glitches usually look like a bunch of random colors flash across the bottom half of the screen. The game I have found it most noticeable on is Snowboard Kids. Here is a video: https://photos.app.goo.gl/kAvNYDFvx1Ush2bP8

bixxewoscht, 2023-06-02:
I heard from people with a N64 HD that it has problems with signal integrity as it is, without using blueretro simultaneously. Reason is most likely the flex-pcb design thats not up to industry standards (ground return path). So likely when using blueretro this amplifies the issue.

nortakales, 2023-06-02:
Yea, I have noticed very infrequently some visual issues, but mostly it is fine. BlueRetro is increasing the frequency by like 100x or 1000x though.

s-taylor, 2023-07-18:
I've ordered the flex cable, I am having issues where the OSD is occasionally inaccessible. Where can I find the installation instructions? I have a friend who is experienced at soldering, hope he can help.

Z3R0XZ, 2023-08-30:
I get graphical glitches, video desync, and signal loss from my N64 Digital via HDMI using Blue Retro. Is this problem related?

[6 more replies]

closed 2023-12-09 by darthcloud