darthcloud/BlueRetro #267: Unable to pull BTRetro BT Mac from Serial Debugging

futurepr0n, 2022-05-02:
I was able to successfully connect to my BlueRetro via Bluetooth and bring up the advanced Config in a browser, and also followed the instructions to use serial connection to monitor what's going on and unfortunately it looks like I am just getting garbled text as response.

[image]

The goal for me is to pair a DualShock 3 to my PS2 - the steps to do this require using a DualShock Pairing utility on windows and also this intermediate step to get the BT Mac from debug. Unfortunately I am struggling to get that info.

[image]

futurepr0n, 2022-05-02:
I was able to pull my BT for it under edge:

[image]

So that looked ok, except after using Sixaxis pair tool and changing the Mac - I cannot get the controller to pair when I start blueretro and the controller up. Is there something I am supposed to do to force pair?

I have a pulsing Blue LED, and solid Red LED

darthcloud, 2022-05-03:
If you connect the controller back on PC and open Sixaxis pair tool again does the current master field show the correct MAC?

futurepr0n, 2022-05-03:
thanks @darthcloud  - I actually had meant to close this issue last night - after more tinkering I believe the issue is my DS3's require a new battery. I think the Battery is so dead that it does not have the power to connect. The pairing is showing that they managed to update what device its paired to. I am waiting on some replacement batteries. Appreciate your help, and thank you for working on such an amazing project!

closed 2022-05-03 by futurepr0n