darthcloud/BlueRetro #269: 3do receiver does not work on the goldstar model

scp-electrocution, 2022-05-03:
I am testing the receiver on a 3do goldstar, the led lights up but the ps4 controller does not pair while on the fz-1 it works perfectly.

darthcloud, 2022-05-04:
If you press EN button on ESP32 after power up does it work?

You could also try to disconnect the 5v from console, and power it via usb.

If none of those work, maybe goldstar one don't have 200mA available at the front port.

scp-electrocution, 2022-05-12:
Pressing the EN key solved the problem, also tested on other 3do models

darthcloud, 2022-05-12:
Get a DS1233A-10+T&R PMIC supervisor and wire it on the EN pin, this will delay esp power up by holding EN 250ms. Also had a 1uF cap & 100uF cap between 3V3 and GND pin of ESP.

See transistor on the back of ESP, this is the PMIC.
[image]

darthcloud, 2022-06-02:
Feel free to reopen if using PMIC do not solve it.

closed 2022-06-02 by darthcloud