darthcloud/BlueRetro #270: Is it posible to pause game when controller battery is low or discharged?

YAGAMI55, 2022-05-04:
So. big thanks to you for BlueRetro)

Is it possible to somehow make the esp32 understand that the gamepad's battery is dead? So that the game is paused or a screen appears with the inscription "connect the gamepad". Recently, I noticed that esp32 does not send any values to the console due to the controller being turned off.

darthcloud, 2022-05-04:
Current no, but in the future I could start pulling battery information for common controllers and maybe make the led flash when its low.

I could maybe also send a "start" button press on BT disconnection.

Demonic098, 2022-05-10:
yep it would be cool to add event scripts
and key combo macros 

> Current no, but in the future I could start pulling battery information for common controllers and maybe make the led flash when its low.
> 
> I could maybe also send a "start" button press on BT disconnection.

egparadigm, 2025-05-09:
> Current no, but in the future I could start pulling battery information for common controllers and maybe make the led flash when its low.
> 
> I could maybe also send a "start" button press on BT disconnection.

I was just wondering about this and yes I think it would be good if the user can set a pause command when the controller disconnects. I nearly got eaten by zombies in RE when my controller suddenly died. It might need to be editable since games don’t always use the same button for pause. 

It is one of the downsides of the product that there’s often no way of knowing the battery status, but this is a separate issue.