darthcloud/BlueRetro #273: Standalone Dreamcast spiff, controller disconnects after about 10min

pceslayer, 2022-05-08:
After playing for about 10 minutes the game pauses and says the controller was disconnected and I am unable to reconnect to Blueretro without a reboot.

Looks like it gets stuck in a reboot loop after the WDT error.

ESP32-WROOM-32D DEVKITC V4 (3rd party) mounted internally
Blueretro 1.5.1 DC spiff

Attached is log
[brdclog.txt](https://github.com/darthcloud/BlueRetro/files/8646226/brdclog.txt)

darthcloud, 2022-05-08:
You got the unused port wired to 3V3 as instructions aay?
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-Cables-Build-Instructions#assembly-instructions-28

pceslayer, 2022-05-08:
All 4 ports are wired.

I have since soldered it directly to the controller pcb. Tried pulling 5v
straight from PSU

Tested port pins to esp32 for connectivity with a multimeter and everything
looked great. Even verified the ground and 5v railes to all be connected.

Also tried with a different WROOM-ESP32D a WROOM-32E and a NodeMCU-ESP32.

darthcloud, 2022-05-08:
Maybe try removing the extra player by wiring them to 3V3 to see if 1P is stable. If stable, rewire one at a time and recheck if stable, this way you can identify which plug is problem.

Which game you use?

pceslayer, 2022-05-08:
I'll give it a try when I get home.

It has happened on every game I've tested so far. Donald Duck Going
Quackers, Dead or Alive 2 and 4x4 Evolution.

I have a GDEMU and USBGDROM... Not sure if they might be doing something
funky polling the controllers or something.

pceslayer, 2022-05-10:
I wired up a different ESP32 with the D0/D1 wired to 3.3v and it was way worse. I couldn't even sync a controller once the Dreamcast booted. It started looping almost as soon as it was powered on. I could pair and hit the web config just fine as soon as I turned off the Dreamcast, or when it was just powered on before it gets past the intro.

So even though these are standalone versions, is still doing some sort of detection at boot? or is that the Dreamcast polling the controller?
[br_dc2.txt](https://github.com/darthcloud/BlueRetro/files/8657528/br_dc2.txt)

pceslayer, 2022-05-12:
I dug out an official ESP32E that I'll try connecting later to see if it works better.

pceslayer, 2022-05-15:
Same issue... disconnects after a few minutes when all soldered up and when the other players are connected to 3.3v it boot loops constantly

pceslayer, 2022-05-24:
Well I don't know what is different about 1.6, but it is working with no drops on the standalone Dreamcast for me.

Played Gigawing and Donald Duck for over about 45min each with no issue.

darthcloud, 2022-05-24:
v1.6 init the wired port a bit sooner, maybe somehow your esp32 was booting in the middle of a data frame and was crashing in loop.

Thanks for taking time to report back!

closed 2022-05-24 by darthcloud

pceslayer, 2022-05-24:
Well it started happening again when I enabled rumble in Demolition Racer - No Exit, but plays fine if I leave it off.

Could it be something with how I have it configured? I have accessory set to both.

darthcloud, 2022-05-24:
Dreamcast rumble support is extremely poorly done, better leave it off until I revise it later.