darthcloud/BlueRetro #277: ESP32 - N64 Memory Pack "RangeError"

Z3R0XZ, 2022-05-17:
Hello, I'm trying to enable CPK functionality of an ESP32 module but the Web interface says "Argh! RangeError: offset is out of bounds!".
I tested with Turok 3 and says the CPK is damaged, sometimes the game repair the CPK, then says is full, is not present or just works for a second then again says is damaged.
I have only a port one connected to the console, the ports 2, 3, 4 (IO05, IO26, IO27) pins are connected to 3.3v coming from the first port and noting.

darthcloud, 2022-05-17:
If you used a phone on web config, try with a PC.

Otherwise try to get the logs:
https://github.com/darthcloud/BlueRetro/wiki/Getting-BlueRetro-debug-logs-via-Serial-port-Windows-10

Save it in a txt file and upload file here.

If you used a clone devkitc maybe try with a genuine devkitc from espressif from mouser or digikey.

Z3R0XZ, 2022-05-17:
Here is the log:
[putty.log](https://github.com/darthcloud/BlueRetro/files/8707369/putty.log)

darthcloud, 2022-05-17:
Did you occur an issue while making that log?, I see no error

Z3R0XZ, 2022-05-17:
Yes, first I did a format, was successful then I try to read the MPK on the web config and the "Range Error" message appear on the web log at 92% more less. Now I need to remove the 3.3v to test on the console when I read and format the CPK.

delislej, 2022-05-20:
i have seen this occasionally as well in the testing of my react version. the code for reading the memory pak is functionally identical. a restart of the blue retro device and or web browser seems to fix it.
[image]

delislej, 2022-05-20:
ok after doing some testing it seems that android has a max byte transfer of 1024 bytes. on the pc the blueretro seems to be sending over 4096 byte chunks, 8 times for 32768 bytes which makes sense. however on the phone, weird stuff is happening, such as this:

`Argh! RangeError: offset is out of bounds
offset: 32454
pakSize: 32768
value byte length:  600
offset: 31854
pakSize: 32768
value byte length:  600
offset: 31254
pakSize: 32768
value byte length:  600
offset: 30654
pakSize: 32768
value byte length:  406
offset: 30248
pakSize: 32768
value byte length:  600
offset: 29648
pakSize: 32768
value byte length:  600
offset: 29048
pakSize: 32768
value byte length:  600
offset: 28448
pakSize: 32768
value byte length:  600
offset: 27848
pakSize: 32768
value byte length:  600
offset: 27248
pakSize: 32768
value byte length:  406
offset: 26842
pakSize: 32768
value byte length:  600
offset: 26242
pakSize: 32768
value byte length:  600
offset: 25642
pakSize: 32768
value byte length:  600
offset: 25042
pakSize: 32768
value byte length:  600
offset: 24442
pakSize: 32768
value byte length:  600
offset: 23842
pakSize: 32768
value byte length:  406
offset: 23436
pakSize: 32768
value byte length:  600
offset: 22836
pakSize: 32768
value byte length:  600
offset: 22236
pakSize: 32768
value byte length:  600
offset: 21636
pakSize: 32768
value byte length:  600
offset: 21036
pakSize: 32768
value byte length:  600
offset: 20436
pakSize: 32768
value byte length:  406
offset: 20030
pakSize: 32768
value byte length:  600
offset: 19430
pakSize: 32768
value byte length:  600
offset: 18830
pakSize: 32768
value byte length:  600
offset: 18230
pakSize: 32768
value byte length:  600
offset: 17630
pakSize: 32768
value byte length:  600
offset: 17030
pakSize: 32768
value byte length:  406
offset: 16624
pakSize: 32768
value byte length:  600
offset: 16024
pakSize: 32768
value byte length:  600
offset: 15424
pakSize: 32768
value byte length:  600
offset: 14824
pakSize: 32768
value byte length:  600
offset: 14224
pakSize: 32768
value byte length:  600
offset: 13624
pakSize: 32768
value byte length:  406
offset: 13218
pakSize: 32768
value byte length:  600
offset: 12618
pakSize: 32768
value byte length:  600
offset: 12018
pakSize: 32768
value byte length:  600
offset: 11418
pakSize: 32768
value byte length:  600
offset: 10818
pakSize: 32768
value byte length:  600
offset: 10218
pakSize: 32768
value byte length:  406
offset: 9812
pakSize: 32768
value byte length:  600
offset: 9212
pakSize: 32768
value byte length:  600
offset: 8612
pakSize: 32768
value byte length:  600
offset: 8012
pakSize: 32768
value byte length:  600
offset: 7412
pakSize: 32768
value byte length:  600
offset: 6812
pakSize: 32768
value byte length:  406
offset: 6406
pakSize: 32768
value byte length:  600
offset: 5806
pakSize: 32768
value byte length:  600
offset: 5206
pakSize: 32768
value byte length:  600
offset: 4606
pakSize: 32768
value byte length:  600
offset: 4006
pakSize: 32768
value byte length:  600
offset: 3406
pakSize: 32768
value byte length:  406
offset: 3000
pakSize: 32768
value byte length:  600
offset: 2400
pakSize: 32768
value byte length:  600
offset: 1800
pakSize: 32768
value byte length:  600
offset: 1200
pakSize: 32768
value byte length:  600
offset: 600
pakSize: 32768
value byte length:  600
reading pak
Init Cfg DOM...
App version: v1.5 n64_spiffs
Reading App version...
Get Api version CHRC...
Getting BlueRetro Service...
Connecting to GATT Server...
Requesting Bluetooth Device...`

the data sent is also completely garbled after a certain point:

pc:
[pc data.txt](https://github.com/darthcloud/BlueRetro/files/8735775/pc.data.txt)

android phone:
[android data.txt](https://github.com/darthcloud/BlueRetro/files/8735776/android.data.txt)

hopefully this helps narrow down where the problem is at.

darthcloud, 2022-05-20:
thanks @delislej for troubleshooting a bit 🥇 

@Z3R0XZ Did you get the error on a phone or a PC?

Z3R0XZ, 2022-05-20:
> thanks @delislej for troubleshooting a bit 1st_place_medal
> 
> @Z3R0XZ Did you get the error on a phone or a PC?

On my phone, my PC Bluetooth that I use to conect PS3 and PS4 controllers don't regonize the ESP32 Module.

darthcloud, 2022-05-25:
Problem is on the first read transaction the ESP send 738 bytes but the javascript function only report receiving 600bytes, after that the offset between javascript and esp get out of sync and we endup with the range error.

Really puzzle why data get lost.

delislej, 2022-05-25:
that is really puzzling. i have no idea why there would be a difference in the data transfer between PC and Android. the javascript code is identical between our versions. for now i'm locking off the n64 controller pak management for mobile.

Z3R0XZ, 2022-05-25:
Thank you I will give a try.

> [v1.6.1_wip.zip](https://github.com/darthcloud/BlueRetro/files/8773711/v1.6.1_wip.zip)
> 
> @Z3R0XZ This firmware should fix the read issue on Android.
> 
> Make sure to delete any previously downloaded mpk file as they are corrupt.
> 
> I tested the write on android and it looked fine to me.

Z3R0XZ, 2022-05-25:
It works! No problem formatting on reading, nor in the game, now is detected normally without any problem on Turok 3.
Thank you so much!

commit 819efad1 (2022-05-26): [BT] Fix memory card read on Android

Closes #277
closed 2022-05-26 by darthcloud