darthcloud/BlueRetro #310: BT controllers stopped working in 1.6.1

Valkristh, 2022-06-17:
Hello, first thanks for the continuous work!

I have installed blueretro in my Wii internally to the 4 gcn ports, the first firmware was 1.2.2 everything working fine, but i updated to 1.6.1 and as the title says all the BT controllers stopped working for some reason idk.
My question is, can i go back or downgrade to 1.2.2 without problems? If the answer is yes I'm gonna try other releases after that one

darthcloud, 2022-06-17:
You can downgrade no problem,

Probably 1.6 broke it? Try 1.6 to confirm that it fail too, try 1.5.1 that one should be fine.

Valkristh, 2022-06-17:
Ok thanks, here is my report
As you said 1.6 broke it and 1.5.1 worked fine

darthcloud, 2022-06-17:
Ok so the fix for #264 likely at fault here

darthcloud, 2022-06-22:
Not sure why but new FW push-pull output doesn't work when there is a small resistance in line ~100ohm

Double check your soldering of the data pin to make sure its not a cold joint. Maybe take a picture too so I can see.

Valkristh, 2022-06-22:
> Not sure why but new FW push-pull output doesn't work when there is a small resistance in line ~100ohm
> 
> Double check your soldering of the data pin to make sure its not a cold joint. Maybe take a picture too so I can see.

Okay I'm gonna take a pic, but (in wii) i remember that there are some components too far (at least 1cm) for a cold joint

Valkristh, 2022-06-24:
[image]

From right to left: gnd, data1, Vin, data 2, data 3, data 4

darthcloud, 2022-06-28:
v1.6.3 should fix this:

https://github.com/darthcloud/BlueRetro/releases/tag/v1.6.3

closed 2022-06-28 by darthcloud