darthcloud/BlueRetro #312: Retro Bit Mega Drive bluetooth not working

micropixel70, 2022-06-18:
I tried a Retrobit Bluetooth brand gamepad for megadrive, but it doesn't work, is there any way to make it work?  thank you

delislej, 2022-06-18:
can we see a picture of your wiring for the ESP32? it's very hard to diagnose anything without it.

micropixel70, 2022-06-19:
> can we see a picture of your wiring for the ESP32? it's very hard to diagnose anything without it.

I don't have the ability to do this, do you know if this brand in bluetooth turns out to work?  thank you

micropixel70, 2022-06-20:
> I ordered the Saturn one, so hopefully it's the same thing and it will get fixed at the same time.

Thanks, I await news💪🏻🙂

closed 2022-07-24 by darthcloud

darthcloud, 2022-07-24:
See https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#retro-bit on how to pair controller.

Only XInput mode is supported (Home + X)

commit b12308c9 (2022-07-24): [BT] Add support for Retro-bit 6 buttons controllers

Fixes #312 & Fixes #313