darthcloud/BlueRetro #313: Controller Profiles?  Retrobit Saturn BT Pad on Genesis, Admiral BT

DariensArcade, 2022-06-19:
Not sure if anyone else is doing this I may have a very peculiar use case or I may have missed the documentation.

I have two shelves of systems and I'm using Auto Detect. (Which doesn't always work but I'll make a 2nd case for that if I can't work it out)  I'm using the super neat AIO boards by PMGDucati. I have 1 "master" esp32 that I then plug in to a db25 "sub" board depending on the system I'm using. (I mention this for a reason.)

Everything has been working amazing with supported controllers _BUT_ I tried a Retrobit Saturn BT controller on my genesis (I've not done my saturn yet since I already have wireless on it) 

The button mapping is off obviously bc BT controllers are 01234, letters are just what we see.  BR is designed around 4 face buttons like ps4/SNES but a retrobit saturn or genesis pad is 6 face.   Specifically (using 240ptest for SegaCD) the retrobit is natively X::A A::B  B::C  Z/L::X  Y::Y C/R::Z   which makes Street Fighter a bit awkward. I can remap using blueretro.io (src: saturn dst:md and move buttons around)...  But I have to do that every time i want to use my 6 button on this system

I couldn't find it in the documentation but is there a way to use/trigger a controller profile using the BT Unique ID?  if I had one blue retro just for the genesis, i could just set and forget it but I'm moving the blueretro around a lot. Using one for (so far) PS2, GC, N64, GEN,.

My ideal setup would be if It detects MD/GEN/SAT and i turn on controller 00:11:22:33:AA:ZZ (whatever the 6 button pad is) it would remap like a "controller profile" , basically pair a deviceid to a set of remaps or have a dropdown where I can see the id/chosen name. 

Not sure where paired IDs reside since Output1 is obviously just whatever you turn on first.

Maybe I want to make submissions to the "Presets" menu?

Thanks,
Darien

darthcloud, 2022-06-19:
Maybe one day for MAC base profile but not in the near future.

I didn't know they made a Bluetooth Saturn ctrl, I ordered one, I'll fix the default mapping when I get it.

DariensArcade, 2022-06-20:
Thanks! Probably just needs like I said a "Profile".

RetroBit does a pretty good saturn controller.  I haven't tried their megadrive one yet (i'm waiting for it to come in red like the wired one). 

I bought one BT (skeleton) and one 2.4Ghz (White) to test the side by side but I just ended up using the BT one since I could quickly go from Saturn to Odroid to PC.

closed 2022-07-24 by darthcloud

darthcloud, 2022-07-24:
See https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#retro-bit on how to pair controller.

Only XInput mode is supported (Home + X)

Use default gamepad preset to reset mapping to default.

commit b12308c9 (2022-07-24): [BT] Add support for Retro-bit 6 buttons controllers

Fixes #312 & Fixes #313