darthcloud/BlueRetro #314: Admiral BT support

DariensArcade, 2022-06-20:
I would say this issue is another profile request but a distinct issue.

The Hyperkin Admiral Premium BT Controller is a smaller style n64 controller available on amazon and in some brick and mortar stores.  When I couldn't find either retrobit or big N's first party 64 controller i bought this one.   It has some wacky features (a dongle that does mempack dumps to SD) and works with 64/Switch/PC.

The controller appears to be pairing  (light goes solid, cannot connect to BR webgui) but no buttons appear to function in any BR system I've tried it with. 

It's recognized in windows/android/switch without the dongle so it is standard BT.    

If there's a way to get a debug log let me know.  

Windows reads as Analog:Analog, Dpad:POV, Start:10, b:1,a:2,l:5,r:6 z:15, C-Left:9, C-up:13, C-Right:12, C-Down:14
there are several overlaps with for example Dualshock4 button assignments. But no function for this contrroller.

Thanks, 
Darien

darthcloud, 2022-06-20:
Added to my amazon list:
https://www.amazon.ca/hz/wishlist/ls/2RMFK6DHX29NQ?ref_=wl_share

If someone buy me one I can investigate to add support.

darthcloud, 2023-01-15:
Support added in v1.8.2
https://github.com/darthcloud/BlueRetro/releases/tag/v1.8.2

closed 2023-01-15 by darthcloud