darthcloud/BlueRetro #323: [PC-FX] Random press in Zeroigar

darthcloud, 2022-07-02:
Random press in Zeroigar

darthcloud, 2023-10-10:
Test build.

This inverted the clock line like the latch and data. This might help keeping the signals in sync within the GPIO matrix.
Switched from SPI mode 0 non-DMA timing to mode 2 DMA ones.

Reduced the SPI setup time to minimum when latch interrupt

[v1.8.4-35-g8fbc9d5_hw1_pcfx.zip](https://github.com/darthcloud/BlueRetro/files/12853208/v1.8.4-35-g8fbc9d5_hw1_pcfx.zip)

DoomyDoomer, 2023-10-10:
This appears to have fixed the issue. I'm going to have it tested further by my tester but seems to work. This also fixed an issue with Team Innocent I was seeing with the character run movement stopping randomly.

darthcloud, 2023-10-15:
Fixed in v1.9 https://github.com/darthcloud/BlueRetro/releases/tag/v1.9

closed 2023-10-15 by darthcloud