darthcloud/BlueRetro #324: gamecube Luigis mansion r2 Button Problem!

jackmiller1986, 2022-07-03:
Hello Guys, 
At first i love this Project. 😉

I have install an esp32 in my gamecube , and have Flash the newest Firmware 1.6.3. 

I have testet with 3 Controllers,  an ps5 , a ps3 , and a joycon remote. 

The ps5 , and ps3 a nice with Connections. When i Connect two joycons,  one of them lost permanently Connecticon. 

But the most expensive Thing ist , i have testet 3 title, Luigis mansion,  mario Sunshine,  and zelda windwaker. 

Mario and windwaker works on both controller All Buttons inkl. The r2 Button.  But only on luigis mansion , IT wont work , on ps5 and ps3 controller , its the only game , who not recocnize the r2 Button,  But all other buttons works in lugis mansion without the r2 thats the only who not works . Is that a Firmware Bug? Because all other games works like a charm . Can anybody confirm that? And why break one joycon the Transmission? 

I hope any body can help.

darthcloud, 2022-07-03:
GC triggers are two buttons in one. an analog part and the digital end. If you use a regular controller and default mapping you only get the analog part.

The end of system manual for GC mention this:
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-System-Specific-User-Manual#mapping-15
```
Using the default mapping, BlueRetro will map the two 
section separately. (Use this if playing with Wii Original 
Classic or TBD Steam controller

Use the "GameCube Merge analog & digital trigger" preset to
 emulate the GameCube trigger using only the analog 
trigger. (Any other type of Bluetooth controller)
```

closed 2022-07-03 by darthcloud

jackmiller1986, 2022-07-03:
IT wont work in luigis mansion all , other games works

darthcloud, 2022-07-24:
Fixed in v1.7.1

commit a34d0792 (2022-07-24): [GC] Add support for trigger 4 bits mode

Fixes #324