darthcloud/BlueRetro #333: Bug Control Ps4 to Ps2

DevDavisNunez, 2022-07-11:
Hello,
I am using the fw psx adapter.
And I use a white CUH-ZCT1U model ps4 1.0 control, first I was using it for a few hours without problems, but then it prevented me from using it, it makes the connection, the control LED turns blue and boom, it disconnects, always like that, I use a control ps4 v1.1 and an 8bitdo snes control and both work without problems, I even load and delete the latest fw to avoid errors, but it still doesn't recognize it, that is, it works for a moment and cuts the connection, I attach the uart log.
[log.txt](https://github.com/darthcloud/BlueRetro/files/9080867/log.txt)

rs1n, 2022-08-25:
I had essentially a similar issue and it was due to a weak battery. In my case, I had a particular controller that would need charging after a few hours of use, and would display the same behavior: it would pair, then immediately power off. After fully recharging the controller, it worked fine.

Armizael, 2023-09-14:
My problem is that the control synchronizes, but when it enters the game it desynchronizes.

darthcloud, 2024-04-20:
@DevDavisNunez Do you still have that issue on latest FW v1.9.2 firmware, I got a couple CUH-ZCT1U and they work fine.