darthcloud/BlueRetro #339: Web bluetooth pairing acting up (DC)

Karabaskarbofos, 2022-07-27:
Hello dear Darthcloud. I'd like to bring up a weird issue. The story so far: I've done a DC internal mod according to this guide: 
[image]
I pair ESP32 to the web server via Bluetooth, set some config and then nothing happens. Esp32  isn't recognised anymore when DC is booted up, only during the intro swirl. Cutting 5v and providing usb power helped, but only with DC power off which interrupts pairing with an error. Is there maybe something wrong with soldering or anything else? Note: controller port board is VA0 Japan
[image]
[image]

Karabaskarbofos, 2022-07-28:
Update: the power may not be enough. Like Yamaha GD-ROM and Samsung GD-ROM, VA0 controller board runs off 3.3V while VA1 consumption is 5V. There's simply not enough voltage 
[image]

darthcloud, 2022-07-28:
I got both VA0 and VA1 Dreamcast working fine personally.

Your soldering job look bad, too long expose wire and residue all over, do it again make sure to strip minimum length wire and cleanup up the board with alcohol after. 

try getting an official DevKit-C (Get it on [Digikey](https://www.digikey.com/en/products/detail/espressif-systems/ESP32-DEVKITC-32E/12091810?s=N4IgTCBcDaIKYGcAOBmMBaAJnAbgawEsAXAY3TThAF0BfIA) or [Mouser](https://www.mouser.com/ProductDetail/Espressif-Systems/ESP32-DevKitC-32E?qs=sGAEpiMZZMv0NwlthflBi3XYxq%252BTapwyHnR2%2FrAf4bY%3D)) 

Moving to discussion